PX4/PX4-Autopilot #15924: Fixedwing Autoland: mismatch between documentation and implementation.

nathanael-k, 2020-10-09:
**Describe the bug**
Parameters for Fixedwing Autoland have a mismatch between documentation and implementation: If we set FW_LND_TLALT to -0.1, the following happens:

`if (land_thrust_lim_alt_relative < 0.0f) {
		land_thrust_lim_alt_relative = 0.66f * _param_fw_lnd_flalt.get();
	}`

**To Reproduce**
Open FixedwingPositionControl.cpp and check line 119

**Expected behavior**
Here is the documentation: https://docs.px4.io/master/en/flying/fixed_wing_landing.html
Quote:

> FW_LND_TLALT | Landing throttle limit altitude (relative landing altitude). The default value of -1.0 lets the system default to applying throttle limiting at 2/3 of the flare altitude.

Since any value below 0.0 triggers the 2/3 behavior, the documentation does not match the code.

**Additional context**
Personally, I would prefer if negative values between (0.0, -1,0) get interpreted as such. This can be useful when you are not that sure about runway elevation, and you want to be conservative. (in that sense, I don't even like the idea that a value like -1.0 triggers a different behavior)
Also, I'll probably be working quite a bit on fixed wing with PX4 in the next few months, how large is the interest in getting changes I make upstream? It would be my first time contributing to such a large project, so I know I'd need some time to know how to best contribute. Personally I think the fixed wing autopilot would benefit of a major overhaul in the direction of much more modularity.

Antiheavy, 2020-10-09:
fyi @Kjkinney

ryanjAA, 2020-10-11:
Definitely interested @nathanael-k Autoland itself could use a serious overhaul and tuning it for repeatable landings was a serious endeavor (since you have the ground to combat...) and there is a lot to be desired and little room for mistakes but overall the process of the landing portion is workable as current but not anywhere as good as it could be. Most likely due to an over simplistic approach to landing. We’ve had conversation about this and about the methodology that would make more sense here: https://discuss.px4.io/t/fixed-wing-auto-landing-brainstorm/9837 and https://github.com/PX4/Firmware/issues/9778 But overall I think any attention to this would be great and I’m sure there are some others here that would agree.

nathanael-k, 2020-10-14:
Thank you very much for the threads, some nice information in there that match my experience / approach to auto land. Ultimately we are investigating auto land for C172 or larger, and PX4 seemed like a good first step. Right now we have some doubts, as the challenges in landing a larger plane just differ strongly to a model airplane, specifically during flare. So right now I am exploring other options.