PX4/PX4-Autopilot #15941: V1.11 auto takeoff high tilt in non-GPS environment

aet2230, 2020-10-11:
Hi.
I am using quadcopter in indoor environment (motion capture), no GPS. Just upgraded to v1.11.0
The issue I encounter: when taking of not from the origin (x=y=0) the drone tends to reach this position during takeoff (high pitch/roll).
[image]
[log_24_2020-9-9-09-22-46.ulg.zip](https://github.com/PX4/Firmware/files/5360603/log_24_2020-9-9-09-22-46.ulg.zip)

It may be observed in the attached screenshot (at t=110 I command takeoff and at t~115 I command land due to high pitch-down) that vehicle position setpoint goes there.
I think the issue comes from the following lines. When I changed them to When I changed it to get “NAN” (as in v1.10) takeoff is ok.
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L317-L318

bresch, 2020-10-14:
related to https://github.com/PX4/Firmware/pull/14543/files

chris-aeviator, 2020-11-22:
@aet2230 could you describe how you get px4 to takeoff in a no-gps environment?

I have set `EKF_MAG_TYPE` to none and `CBRK_GPS` to 240024 but px4 refuses to takeoff due to missing position lock

aet2230, 2020-11-22:
Px4 must have other than GPS positioning then. We use motion capture as positioning system, the data is sent via mavros
Set EKF_AID_MASK 24 (vision position and yaw fusion)
EKF_HGT_MODE to 3 (vision)

chris-aeviator, 2020-11-22:
I'm planning to use a machine learning approach to estimate hight and command a trajectory on every frame. From my understanding I need to switch out attitude estimators. First step though would be to approach it step by step, therefore I'm looking for a way to just hover (baro height) tethered and command velocity bodies/ attitude vectors.

aet2230, 2020-11-22:
Can't help you then. AUTO.TAKEOFF is one of Auto mode, which requires position lock. You would suggest you to try to use Offboard mode to create your own takeoff sequence

chris-aeviator, 2020-11-22:
Now I get the point, thanks for making this hint - I wasn't even trying to activate Offboard when on ground :+1:

DogukanAltay, 2021-01-14:
@aet2230 can you tellme which mavros topic you publish for moCap as positioning system and what is the rate of it? Thanks!

aet2230, 2021-01-15:
/mavros/vision_pose/pose

DogukanAltay, 2021-01-16:
@aet2230 Thank you so much for the reply. I am literally doing the same thing with you but somehow the external position signal does not consumed by the mavros neither in SITL, HITL or the drone setup. One last question is what is your mavros, mavlink and px4 version that you use in your setup.

aet2230, 2021-01-17:
verify you put correct timestamp to your vision position message and set  
EKF_AID_MASK 24 (vision position and yaw fusion)
EKF_HGT_MODE to 3 (vision)

we use old mavros version 0.21.5
but I think it should work with newer ones