PX4/PX4-Autopilot #15942: V1.11 - vision yaw fusion

aet2230, 2020-10-11:
Hi.
I am using quadcopter in indoor environment (motion capture). Just upgraded to v1.11.0
The issue I encounter: when drone is powered on or rebooted facing southbound, it refuses to receive vision yaw (heading rotates 360 deg and takeoff is rejected due to yaw estimate error).
In v1.10 it worked well
I use the following parameters :
EKF2_AID_MASK	24 (vision position and yaw fusion)
EKF2_HGT_MODE 3 (vision height fusion)
EKF2_MAG_TYPE 5 (none)
Something I have to change?
Thanks

mhkabir, 2020-10-11:
Known ecl bug: https://github.com/PX4/ecl/pull/807

Adam-Laidlaw, 2020-11-10:
I was having a similar issue in 1.11, when using EKF2_MAG_TYPE = 5 (none) my ev data was not being used to change the heading- I think I found the problem:

In controlMagFusion() there's a check for if MAG_FUSE_TYPE_NONE, and then noOtherYawAidingThanMag(), even if you're sending good vision data on boot this will be run and _is_yaw_fusion_inhibited is set to true.

The problem is that when EV data is being sent after this, the yaw fusion inhibited flag is never set back to false.

I made a quick change, inside the block where MAG_FUSE_TYPE_NONE is checked, I added an else where _control-status.flags.ev_yaw is checked, if true then _is_yaw_fusion_inhibited is set to false.

This works well for me in SITL flight testing, robust to intermittent vision data.

aet2230, 2020-11-11:
Thanks @AdamLink1270 !
After your comment I looked at controlMagFusion(). I saw that if EKF2_MAG_TYPE  = 5, EKF still has an option to fuse magnometer and this causes the problem you explained and solved usin "else"
I payed attention that at the beginning of controlMagFusion(), it checks if 
	// handle undefined behaviour
	if (_params.mag_fusion_type > MAG_FUSE_TYPE_NONE) {
		return;
	}

So I put EKF2_MAG_TYPE = 6 (undefined), and got the same behavior. Now EKF **REALLY** does not fuse MAGnometer. This solved the problem for me (I do'nt want to use magnometef fusion at all)

closed 2026-04-10 by github-actions[bot]