PX4/PX4-Autopilot #15944: Durandal can't connect to companion computer

bys1123, 2020-10-12:
**Describe the bug**
Durandal can't connect to companion computer, tested on Telem2.
Same setting on Pixhawk4 works fine

**Log Files and Screenshots**
[image]
[image]

dagar, 2020-10-18:
Do you get any particular error? What does `mavlink status` on the console say?

bys1123, 2020-10-18:
[image]

dagar, 2020-10-18:
On the Durandal side what does `mavlink status` show?

bys1123, 2020-11-17:
I don't have a Durandal on hand recently, this output are from my client, hope it will help, let me know if you need further information
[image]

bys1123, 2020-11-18:
Other users tested fine on latest firmware, closed.

closed 2020-11-18 by bys1123