PX4/PX4-Autopilot #15951: HITL Battery Missing Parameters

McKayCyborg, 2020-10-13:
It appears that this issue is still present in the current PX4 firmware build. I am trying to run HITL using AirSim on a Pixhawk 4 board and I am unable to run a mission due to the following errors:

When I boot up, it gets this issue with battery parameters:
[image]

Then, when trying to upload a mission, it gets these issues:
[image]

[image]

Also, I get a message saying that the vehicle requires setup prior to flight.
[image]

Steps to reproduce:
1) Flash Pixhawk with latest firmware
2) Set airframe to HIL Quadcopter X
3) Calibrate RC radio and set flight modes
4) Try to upload a waypoint mission with AirSim 

Current setup:
QGC Version - v4.0.10
PX4 Firmware version - v1.11.1
Airframe - HIL Quadcopter X

What I have tried:
1) Re-flashing the board
2) Re-installing QGC
3) Resetting all parameters on the board
4) Setting the CBRK_SUPPLY_CHK to 894281

Nothing appears to work and I am unable to fly a mission. I am able to arm and fly it manually using an RC Radio controller, however.

Has this issue actually been fixed?

Jaeyoung-Lim, 2020-10-13:
@bperseghetti Any insights?

McKayCyborg, 2020-10-13:
After some looking around, I decided to try flashing an older version of PX4 on the Pixhawk to see if that works. Using v1.9.2 I have managed to successfully upload missions and fly them autonomously in AirSim. Is there a chance that something was fixed and then reverted?

I did notice that on the first board I tried it worked straight away with v1.9.2 and on another board I needed to swap out a perfectly functional SD card for another. Is it possible that the SD card needs to be wiped or re-formatted before uploading missions with HITL?

bperseghetti, 2020-10-14:
I had this exact issue a month or so ago, the issue was with cache in QGC and the params saved on the FCU. I had the issue though in Ubuntu using Gazebo. My solution for that was removing QGC and clearing the associated cache: 
`rm -rf ~/.cache QG*`
Clear the Pixhawk by connecting to it using `./Tools/mavlink_shell.py /dev/ttyACM0`.
Press enter a few times till you see the nuttx shell (nsh>) pop up and then run `mtd erase`.

It should look like:
```
nsh> mtd erase
Erasing /fs/mtd_params
Erased 16384 bytes
Erasing /fs/mtd_waypoints
Erased 16384 bytes
nsh>
```

McKayCyborg, 2020-10-15:
@bperseghetti Thanks for the reply!

Unfortunately, this didn't seem to work for me. I couldn't find the QGroundControl cache on my windows machine (I don't know where to look or what it's called), so this is what I did:

1) Opened 'Add or Remove Programs' and uninstalled QGroundControl. Verified that the program was gone.
2) Ran the installer again, and installed QGC v4.0.10
3) Re-flashed my Pixhawk 4 with firmware v1.11.1 and went through the setup process again (radio, flight modes etc..). I also set the CBRK_SUPPLY_CHK to 894281.
4) Opened the NSH console using the mavlink console in QGC and ran 'mtd erase' and received exactly what you received in your reply.
5) Booted up AirSim, connected to the vehicle over serial, and it connected to QGC immediately, at which point I received a few of the 'missing battery status params' messages
6) Opened the mission planner screen, at which point I received a few 'maximum retries exceeded' messages
7) Attempted to upload a mission and received the 'GeoFence transfer failed. Retry transfer. Error: Mission write failed, vehicle failed to send final ack' and 'Rally Point transfer failed. Retry transfer. Error: Mission write failed, vehicle failed to send final ack' messages
8) Checked that manual mode still works, which it does
9) Checked the MAVLink Link Status Loss rate which is 0%

Somehow the vehicle is still refusing to upload these missions.

McKayCyborg, 2020-10-15:
I think I managed to find the QGC cache located in AppData/roaming and AppData/local and it was called QGroundControl.org. I deleted this folder in both of those locations, reinstalled QGC v4.0.10 and ran the mtd erase command on the pixhawk with v1.11.1 and am still receiving all of the same error messages, although now I'm also receiving this error:

[image]

as well as the usual:

[image]

[image]

bperseghetti, 2020-10-15:
Want to move debugging this to slack?

McKayCyborg, 2020-10-15:
@bperseghetti thanks, yeah I've just moved it over to the slack.

McKayCyborg, 2020-10-26:
I managed to "fix" this issue by flashing ArduPilot on the Pixhawk, and then re-flashing PX4 v1.9.2. Somehow, this gave me a clean slate to work with and everything appeared to work fine. Mind you, this was on v1.9.2 and I haven't yet tested that this is okay on v1.11.1 (latest release), but I have a feeling that it will be fine on that too.

closed 2020-10-26 by McKayCyborg

dagar, 2020-10-26:
> I managed to "fix" this issue by flashing ArduPilot on the Pixhawk, and then re-flashing PX4 v1.9.2. Somehow, this gave me a clean slate to work with and everything appeared to work fine. Mind you, [...]

What that does is fully wipe the parameters.