PX4/PX4-Autopilot #15953: UAVCAN servo sending out periodic messages not visible in node overview

Etruscian, 2020-10-13:
**Describe the bug**
UAVCAN servo sending out periodic messages not visible in node overview

**To Reproduce**
Steps to reproduce the behavior:
1. Attach UAVCAN servo to can connector
2. Turn on servo with external power supply
3. Turn on px4 autopilot
4. open mavlink console and type `uavcan status`

**Expected behavior**
Eventhough PX4 does not support servos over UAVCAN, the underlying library does. Since the servo is sending periodic status messages, I would expect the servo to show up in the node status which is provided by the underlying library.

**Pictures**
[image]
[image]

Note that for these pictures, a VESC was also connected to the uavcan bus on port 1, while the uavcan servo was connected to the canbus on port 2.

**Drone (please complete the following information):**
- pixhawk 2 autopilot
- HITEC MD950TW UAVCAN servo

dagar, 2020-10-13:
Do you have a Zubax Babel or something you can use to inspect the CAN bus traffic?

Note, here's UAVCAN servo support that needs to be resurrected. https://github.com/PX4/Firmware/pull/12984

Etruscian, 2020-10-13:
I currently do not have a CAN bus device that I can use to inspect the CAN bus but i will shortly. 

I did see that pull request. It is however very much out of sync with the current master (especially the mixer part), so while I did use it as an inspiration, I decided not to try to convert that pr to the current master due to inexperience with both the mixer and uavcan. However, I might go back on that decision based on the knowledge I get from trying to understand it all.

Etruscian, 2020-10-15:
I haven't connected the CAN bus inspector yet, but I decided to see if Ardupilot could connect with the servos (they state that they do have uavcan servo support). While the SLCAN connection does not show the servo (it does show the esc), when clicking on the inspector, I can see a `uavcan::equipment::actuator::status` typed message coming in with valid values at the interval that I specified with the configurator.

This could indicate a bug on the side of uavcan or px4...