PX4/PX4-Autopilot #15962: make px4_sitl gazebo_plane error

shu-bamma, 2020-10-14:
I am using ubuntu 16.04 and gazebo 7. After building the jmavsim simulation i tried to build it  with gazebo. This is the command i used:
`shubham@haha:~/src/Firmware$ make px4_sitl gazebo_plane`
This got me the following error:
```
[1/4] Performing build step for 'sitl_gazebo'
ninja: error: '/home/shubham/src/Firmware/Tools/sitl_gazebo/PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND', needed by 'Airspeed.pb.cc', missing and no known rule to make it
FAILED: cd /home/shubham/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build /home/shubham/src/Firmware/build/px4_sitl_default/build_gazebo
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
```
Any help will be appreciated.

Jaeyoung-Lim, 2020-12-06:
`apt-get install -y protobuf-compiler`  Should be the right way to install the protobuf compiler.

Have you tried running `make clean` and try building again?

shu-bamma, 2020-12-07:
> `apt-get install -y protobuf-compiler` Should be the right way to install the protobuf compiler.
> 
> Have you tried running `make clean` and try building again?

yeah this worked thanks.

closed 2020-12-07 by shu-bamma