PX4/PX4-Autopilot #15963: Unable to build stable Firmware

JAPolo2, 2020-10-14:
**Describe the bug**
Unable to do calibration on QGC using a custom firmware

**To Reproduce**
1. Clone the PX4 firmware from git using Ubuntu 20.04 (_git clone https://github.com/PX4/Firmware.git --recursive_)
2. Make a clean build (without modifying the code) for getting Cube Orange firmware
3. Flash a Orange Cube with the custom firmware using QGC daily version on Windows 10
4. Try to calibrate sensors. Gyroscope calibration never completes (progress bar still loading); sometimes, compass and airframe don't follow the configured autopilot orientation.
5. Try to arm without success.

**Expected behavior**
To be able to build a stable custom firmware and to be able to calibrate and arm the drone with no problem.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/master/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

**Screenshots**

- When doing a clean build, PX4 version is v1.11.0-rc3 instead of v1.11.1 (latest version of stable release)
[image]

[image]

-Incomplete gyroscope calibration
[image]

**Drone:**
- Custom Y tricopter (1 main front motor with two rear motors)

**Additional context**
- Clean builds were made without any modification in the code.
- The objective of having a custom firmware is to add a new airframe configuration.

dagar, 2020-10-14:
Is there a specific error? Can you reproduce with a log from boot (SDLOG_MODE 2)?

JAPolo2, 2020-10-15:
> Is there a specific error? Can you reproduce with a log from boot (SDLOG_MODE 2)?

Actually, I'm a beginner using PX4 so I don't know exactly how I can get a log from boost...Could you please show me how to do it?
Anyway, I solved the problem by changing the Firmware release version:

`git clone https://github.com/PX4/Firmware.git` (no `--recursive`)
`cd Firmware/`
`git checkout v1.11.1`

After performing a build for Cube orange and flashing the controller with QGC, I get the latest firmware version and I'm able to do the sensor calibration and arming the drone:

[image]

I don't know if I'm the only one to have that problem, but it's weird that I'm not able to build the code by just cloning the latest firmware version. I mean, if I don't use `git checkout` for choosing v1.11.1, the firmware I build is unstable...

julianoes, 2023-04-03:
Sounds like the problem was solved for OP, so I'll close this.

closed 2023-04-03 by julianoes