PX4/PX4-Autopilot #15969: Disable blending vtol transition problem

moreba1, 2020-10-14:
HI

Blending Airspeed

Parameter: VT_ARSP_BLEND

"By default, as the airspeed gets close to the transition speed, multirotor attitude control will be reduced and fixed wing control will start increasing continuously until the transition occurs.
Disable blending by setting this parameter to 0 which will keep full multirotor control and zero fixed wing control until the transition occurs."

I set VT_ARSP_BLEND =0  but blending started and did not stoped.

moreba1, 2020-10-18:
@dagar 
@RomanBapst

ThomasRigi, 2020-10-28:
As a quick fix you can set VT_ARSP_BLEND to your transition airspeed

moreba1, 2020-10-28:
> 
> 
> As a quick fix you can set VT_ARSP_BLEND to your transition airspeed

I setted VT_ARSP_BLEND to transition airspeed but it did not stop.

moreba1, 2020-10-28:
@RomanBapst

sfuhrer, 2021-01-18:
@moreba1 Disabling fixed-wing actuation during transition is currently not possible. It will be active from the start of the transition. The blending only refers to the MC actuaion (it will "blend it out"). --> We need to fix the parameter description  ~"fixed-wing control will start increasing continuously until the transition occurs."~ In some way, the FW actuators are also "blended in" automatically, as they gain effectiveness with increased airspeed.

May I know why you want it to be disabled during transition?

moreba1, 2021-01-23:
> May I know why you want it to be disabled during transition?

For prevention from Intermingling of the controllers.

sfuhrer, 2021-01-24:
> For prevention from Intermingling of the controllers.

With the current VTOL architecture where we have separate controllers running for MC and FW flight, we always need some sort of blending from one to the other.

closed 2021-01-24 by moreba1