PX4/PX4-Autopilot #15981: Missing MPC_xx parameters

JAPolo2, 2020-10-15:
**Describe the bug**
I'm missing almost all the MPC_xx parameters when adding a new airframe using PX4 v1.11.1.

**To Reproduce**
Steps to reproduce the behavior:
1. Adding a new airframe following the steps giving by the dev guide (https://dev.px4.io/master/en/airframes/adding_a_new_frame.html) 
2. Make a clean build in Ubuntu 20.04
3. Upload the custom firmware using QGC on Windows 10
4. Check the Tuning section of QGC

**Expected behavior**
To be able to modify the Tuning setup of the custom airframe without errors

**Log Files and Screenshots**
[image]

**Drone (please complete the following information):**
- Custom Tricopter Y
- Controller: CubePilot - Cube Orange

dagar, 2020-10-18:
What does your custom airframe look like? I'm wondering if some of the normal modules aren't running (eg mc_pos_control). Through the console you could check `work_queue status` to see if they're running.

julianoes, 2023-04-03:
There was no response, so I'm closing this.

julianoes, 2023-04-03:
Related to https://github.com/PX4/PX4-Autopilot/issues/15591.

closed 2023-04-03 by julianoes