PX4/PX4-Autopilot #15982: FAILED: px4_io-v2_default.elf

Amith4504, 2020-10-16:
`-- PX4 version: v1.11.0-rc3-504-g7d6f8dc882
-- PX4 config file: /home/amith/Documents/DRONES/PX4/v1.11.0/Firmware/boards/px4/fmu-v3/default.cmake
-- PX4 config: px4_fmu-v3_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.2.1
-- The C compiler identification is GNU 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working C compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /home/amith/anaconda3/bin/python3 (found suitable version "3.7.4", minimum required is "3") 
-- build type is MinSizeRel
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-456-gf666ebb
-- Building and including px4_io-v2_default
-- Found PythonInterp: /home/amith/anaconda3/bin/python3 (found version "3.7.4") 
-- Using C++11
-- Release build type: MinSizeRel
-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- ROMFS: Adding rc.board_mavlink
-- Found SVD: 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/amith/Documents/DRONES/PX4/v1.11.0/Firmware/build/px4_fmu-v3_default
[0/1367] git submodule platforms/nuttx/NuttX/nuttx
[3/1367] git submodule platforms/nuttx/NuttX/apps
[7/1367] git submodule src/lib/ecl
[16/1367] git submodule src/drivers/gps/devices
[20/1367] git submodule src/drivers/uavcan/libuavcan
[22/1367] git submodule mavlink/include/mavlink/v2.0
[1350/1367] Performing configure step for 'px4io_firmware'
-- PX4 version: v1.11.0-rc3-504-g7d6f8dc882
-- PX4 config file: /home/amith/Documents/DRONES/PX4/v1.11.0/Firmware/boards/px4/io-v2/default.cmake
-- PX4 config: px4_io-v2_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.2.1
-- The C compiler identification is GNU 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working C compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /home/amith/anaconda3/bin/python3 (found suitable version "3.7.4", minimum required is "3") 
-- build type is MinSizeRel
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-456-gf666ebb
-- Configuring done
-- Generating done
-- Build files have been written to: /home/amith/Documents/DRONES/PX4/v1.11.0/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware
[1350/1367] Performing build step for 'px4io_firmware'
[0/227] git submodule platforms/nuttx/NuttX/nuttx
[1/227] git submodule platforms/nuttx/NuttX/apps
[225/227] Linking CXX executable px4_io-v2_default.elf
FAILED: px4_io-v2_default.elf 
: && /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++  -mcpu=cortex-m3 -mthumb -Os -DNDEBUG  --specs=nosys.specs platforms/nuttx/CMakeFiles/px4.dir/__/common/empty.c.obj  -o px4_io-v2_default.elf -L/home/amith/Documents/DRONES/PX4/v1.11.0/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/external/Install/lib -nostartfiles -nodefaultlibs -nostdlib -nostdinc++ -fno-exceptions -fno-rtti -Wl,--script=/home/amith/Documents/DRONES/PX4/v1.11.0/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware/NuttX/nuttx-config/scripts/script.ld -Wl,-Map=px4_io-v2_default.map -Wl,--warn-common -Wl,--gc-sections -Wl,--start-group NuttX/apps/libapps.a NuttX/nuttx/arch/arm/src/libarch.a NuttX/nuttx/binfmt/libbinfmt.a NuttX/nuttx/libs/libc/libc.a NuttX/nuttx/boards/libboards.a NuttX/nuttx/libs/libxx/libxx.a NuttX/nuttx/drivers/libdrivers.a NuttX/nuttx/fs/libfs.a NuttX/nuttx/mm/libmm.a NuttX/nuttx/sched/libsched.a -Wl,--end-group -lm -lgcc -Wl,--print-memory-usage src/modules/px4iofirmware/libpx4iofirmware.a NuttX/apps/libapps.a platforms/nuttx/src/px4/stm/stm32f1/io_pins/libarch_io_pins.a src/lib/mixer/libmixer.a src/lib/mixer/HelicopterMixer/libHelicopterMixer.a src/lib/mixer/MultirotorMixer/libMultirotorMixer.a src/lib/mixer/NullMixer/libNullMixer.a src/lib/mixer/SimpleMixer/libSimpleMixer.a src/lib/mixer/MixerBase/libMixerBase.a src/lib/rc/librc.a src/lib/output_limit/liboutput_limit.a NuttX/nuttx/arch/arm/src/libarch.a boards/px4/io-v2/src/libdrivers_board.a platforms/nuttx/src/px4/stm/stm32f1/hrt/libarch_hrt.a src/lib/parameters/libparameters.a src/lib/parameters/flashparams/libflashparams.a NuttX/nuttx/arch/arm/src/libarch.a boards/px4/io-v2/src/libdrivers_board.a platforms/nuttx/src/px4/stm/stm32f1/hrt/libarch_hrt.a src/lib/parameters/libparameters.a src/lib/parameters/flashparams/libflashparams.a NuttX/nuttx/libs/libxx/libxx.a NuttX/nuttx/libs/libc/libc.a NuttX/nuttx/drivers/libdrivers.a NuttX/nuttx/fs/libfs.a NuttX/nuttx/mm/libmm.a NuttX/nuttx/sched/libsched.a -lgcc src/lib/parameters/tinybson/libtinybson.a platforms/common/libpx4_platform.a msg/libuorb_msgs.a -lm && :
Memory region         Used Size  Region Size  %age Used
           flash:       56392 B        60 KB     91.78%
            sram:        3832 B         8 KB     46.78/opt/gcc-arm-none-eabi-7-2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/libm.a(lib_a-wf_sqrt.o): In function `sqrtf':
wf_sqrt.c:(.text.sqrtf+0x3e): undefined reference to `__aeabi_f2d'
wf_sqrt.c:(.text.sqrtf+0x54): undefined reference to `__aeabi_ddiv'
wf_sqrt.c:(.text.sqrtf+0x70): undefined reference to `__aeabi_d2f'
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/px4io_firmware/px4io_firmware-build 
cd /home/amith/Documents/DRONES/PX4/v1.11.0/Firmware/build/px4_fmu-v3_default/external/Build/px4io_firmware && /usr/local/bin/cmake --build .
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_fmu-v3' failed
make: *** [px4_fmu-v3] Error 1`

### Failed to make px4_fmu-v3_default
**arm-none-eabi-gcc --version** arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]

Firmware version :  v1.11.0
Ubuntu : 18.04
This GCC seems to be incapable of building firmware for Pixhawk Cube.

@dagar Can you suggest a quick fix for this ?

taileron, 2020-10-16:
For men the same there is a fixed date in September before that GCC 7.2.1 works GCC 9.3.1 is needed afterwards

closed 2020-10-19 by dagar

taileron, 2020-10-19:
@dagar the build procedure now completely works again with GCC 7.2.1 so it´s still the most recommended version ?
(I had the feeling that with GCC 9.3.1 compiled code flies not as good as with GCC 7.2.1 ...)

dagar, 2020-10-19:
> @dagar the build procedure now completely works again with GCC 7.2.1 so it´s still the most recommended version ?
> (I had the feeling that with GCC 9.3.1 compiled code flies not as good as with GCC 7.2.1 ...)

The recommended version is to always to match CI, docker, setup scripts (https://github.com/PX4/Firmware/blob/master/Tools/setup/ubuntu.sh), etc. That's currently arm-none-eabi-gcc 9.3.1 for master and 7.2.1 for PX4 v.1.11.1. I haven't seen any evidence of regression when we upgraded from GCC 7 to 9, and in fact the binary size decreased with the upgrade (better code generation).

taileron, 2020-10-21:
@dagar One more thing I found out: as soon as GCC 9.3.1 is used, some mixers can't be loaded into the IO as long as PX4IO_MAX_MIXER_LENGTH=330 is set in mixer.h. Even with 430 my mixer with 5 entries in the output doesn't work anymore.
Using GCC 7.2.1 all mixers load correctly also in the current master.

dagar, 2020-10-21:
> @dagar One more thing I found out: as soon as GCC 9.3.1 is used, some mixers can't be loaded into the IO as long as PX4IO_MAX_MIXER_LENGTH=330 is set in mixer.h. Even with 430 my mixer with 5 [...]

That's fairly significant actually. Let's get a new issue going for that specifically and dig into it.