PX4/PX4-Autopilot #15984: PX4IO ERROR and ERROR Tune upon selecting Blade130 Airframe

Amith4504, 2020-10-16:
I uploaded the built master firmware onto CUBE ORANGE and Cube Black . Flashing worked fine . Further upon selecting **Blade130 mixe**r in QGC , i am getting
 https://docs.px4.io/master/en/getting_started/tunes.html#error-tune   and 
https://docs.px4.io/master/en/getting_started/tunes.html#program-px4io-fail 
ERROR TONE . There seems to be bug just with Blade130 mixer configuration. 
The error does not happen with any other mixer configs .

dagar, 2020-10-18:
Are you getting the same error on both after a fresh power cycle? If you try generic quadcopter (SYS_AUTOSTART 4001) do they work?

Amith4504, 2020-10-18:
Error Tune does not occur with any other airframes / default SYS_AUTOSTART . But upon selecting blade 130 helicopter mixer in QGC , error tune starts upon rebooting which is required after selecting new airframe. 
The error tune occurs just with Blade 130 mixer.

max11gen, 2021-03-12:
Has this been investigated in any way? I have the same issue.

bkueng, 2022-10-25:
There were a lot of changes in IO and mixer, so I'm closing this. Reopen if still an issue.

closed 2022-10-25 by bkueng