PX4/PX4-Autopilot #15988: DMA allocation

BazookaJoe1900, 2020-10-17:
**Describe the bug**
I am not sure but it possible that there is some error on the DMA allocation on board FMUv5:
on the defconfig the next DMA are set to on:
183: CONFIG_STM32F7_SPI1_DMA=y 
218: CONFIG_UART8_RXDMA=y 
225: CONFIG_USART2_RXDMA=y 
229: CONFIG_USART3_RXDMA=y 
231: CONFIG_USART3_TXDMA=y 
234: CONFIG_USART6_RXDMA=y 

https://github.com/PX4/Firmware/blob/224be8ba2473d334466316a9dee1bdaf332ca90d/boards/px4/fmu-v5/nuttx-config/nsh/defconfig#L181

https://github.com/PX4/Firmware/blob/224be8ba2473d334466316a9dee1bdaf332ca90d/boards/px4/fmu-v5/nuttx-config/nsh/defconfig#L208
but on the DMA table, there written that also 

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/master/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

dagar, 2020-10-17:
The table (documentation) is out of date.

BazookaJoe1900, 2020-10-17:
I will update it, there are some more data that can be filled

BazookaJoe1900, 2020-10-17:
moved to PR: fmu-v5: fix DMAmap #15989

closed 2020-10-17 by BazookaJoe1900