PX4/PX4-Autopilot #15991: Intermittent issue of mission execution in v1.11.1

aman151294, 2020-10-19:
On giving mission the UAV takes off to the mission height and keeps hovering at the mission height.

GCS used: QGC v3.5.6 (Stable Release)
px4 Firmware: v1.11.1

Detailed Observations:
We have observed that flying the already uploaded mission for the second flight (i.e. same mission without re-uploading is flown again and again) the UAV is taking off at its home position at given mission height and does not go on the next waypoint instead it simply hover at home position, however the mission mode was still showing MISSION on GCS. 

After more than 50 flights, we have concluded that this behavior is intermittent/random in nature and we are unable to debug the reason for this. 

However, if the vehicle is rebooted or the mission is re-uploaded after every flight, then we don't face this issue.

Need help in figuring out the reason and also if anyone else is also facing issues in multiple mission executions without re-uploading or rebooting the UAV.

Thanks

dagar, 2020-10-19:
Do you have a log you can share that hit the problem? https://logs.px4.io/

aman151294, 2020-10-20:
Hi, 

Mentioned below is the link of uploaded log regarding this issue:

https://review.px4.io/plot_app?log=c6112cdf-d379-4b5c-bc15-851d7843c746

The mission that should have been executed is similar to the one executed in this log:

https://review.px4.io/plot_app?log=104ed141-efee-4aca-aea2-1999e82935f4

Please let me know if you require anything else regarding this.