PX4/PX4-Autopilot #15993: Disabling EKF in no global position

moreba1, 2020-10-19:
Describe problem solved by the proposed feature :
In EKF fail condition and "no global position" because of the mag or other reasons, the drone can not use GPS position.

Describe your preferred solution :

There is must be possible to disable EKF and use GPS and barometer for landing without using EKF.

dagar, 2020-10-20:
There's emergency descend landing with loss of position estimate. What's the specific problem you're trying to solve?

In terms of decreasing the likelihood of fully losing a position estimate multi-EKF (https://github.com/PX4/Firmware/pull/14650) should be available in master for testing soon.

moreba1, 2020-10-20:
Drone in descend landing does not control position and move with wind while gps position is available.

dagar, 2020-10-20:
I would dig into why the existing estimator is failing rather than expecting something else to takeover using that same (presumably flawed) data. If you have logs to share (https://logs.px4.io/) I can take a quick look.

The multicopter position controller only requires a valid local (not global) position estimate to function. So for example additionally using optical flow or a VIO system (eg Intel Realsense T265) can help.

If you're having magnetometer issues then look at (or be aware) of the backup yaw estimator in EKF2 (https://youtu.be/CwehBx2hJcg?t=491). If your mag doesn't work well due to proximity to high current then I'd try power compensation (http://docs.px4.io/master/en/advanced_config/compass_power_compensation.html). Having multiple mags and using multi-EKF (#14650) could also help if you have separated magnetometers.

moreba1, 2020-10-20:
https://review.px4.io/plot_app?log=bc09e776-838b-4b82-a435-f6b3fcd8a375

> The multicopter position controller only requires a valid local (not global) position estimate to function.

In this log, VTOL does not control position during descending but GPS position is available.

dagar, 2020-10-21:
> https://review.px4.io/plot_app?log=bc09e776-838b-4b82-a435-f6b3fcd8a375
> 
> > The multicopter position controller only requires a valid local (not global) position estimate to function.
> 
> In this log, VTOL does not control position during descending but GPS position is available. reason of the problem was the low distance of compass to wires but "no global position" may occur for various and simple reasons such as vibration for everyone.

[image]

Your magnetometer data is incredibly bad, I would consider relocating the external compass and upgrade to at least PX4 v1.11 with the backup yaw estimator to recover in these situations rather than losing the position estimate.

closed 2020-10-21 by moreba1