PX4/PX4-Autopilot #16001: px4io issues to review

dagar, 2020-10-20:
- spektrum bind not working
     - vehicle_command ack
 - helicopter mixer failing (blade130 `SYS_AUTOSTART 16001`)

LorenzMeier, 2020-10-20:
I'm digging into the heli right now.

MaEtUgR, 2021-02-04:
Spektrum bind was covered by:
https://github.com/PX4/PX4-Autopilot/pull/16466
https://github.com/PX4/PX4-Autopilot/pull/16470
https://github.com/PX4/PX4-Autopilot/commit/0bdae5745aec4efb2a32e0967ae43ba0563852b3

liganggg, 2022-08-29:
Is there any solution to this problem, I use cuav v5+ whitch contain io board,   it report same error about mixer failing

ERROR [px4io] mixer send error -1
ERROR [mixer] failed to load mixers from /etc/mixers/blade130.main.mix
ERROR [mixer] failed to load mixer
ERROR  [init] Failed loading mixer: /etc/mixers/blade130.main.mix

and then i set param SYS_USE_IO false, it looks like work normal. 
Can anyone give some advice?　thanks

MaEtUgR, 2022-08-29:
@liganggg We're flying multiple helis recently. I suggest using the latest main branch (or even https://github.com/PX4/PX4-Autopilot/pull/20114 which will be merged soon) and configuring your helicopter actuators using dynamic control allocation:
https://docs.px4.io/main/en/config/actuators.html (latest QGC daily for the UI support)
This is the way supported from now on, much easier and should not have the problem with the IO that you were facing before.
Feel free to ask further questions on your way.

bkueng, 2022-10-25:
Closing as mixers got removed

closed 2022-10-25 by bkueng