PX4/PX4-Autopilot #16006: px4_raspberrypi 1.11.0 with Navio2

blazczak, 2020-10-21:
**Describe the bug**
Preflight keeps failing with `Fail: Gyros inconsistent - Check Cal`. Re-calibrating doesn’t solve it, the issue comes back again later. There were a bunch of timeouts for sensors throughout - nothing like that happens with 1.10.1 and same hardware. A clear blocker.

**To Reproduce**
Steps to reproduce the behavior:
Build 1.11.0 PX4 on Ubuntu 20, use instructions in https://community.emlid.com/t/navio2-support-for-ubuntu-20-04/19956 to create the spidev stuff if needed.

**Expected behavior**
Expect the calibration to "stick" when initially (properly, successfully) completed.

**Log Files and Screenshots**
```
px4 starting.

INFO  [px4] Calling startup script: /bin/sh posix-configs/rpi/px4.config 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
INFO  [linux_pwm_out] Loaded mixer from file ROMFS/px4fmu_common/mixers/quad_x.main.mix
INFO  [linux_pwm_out] Starting PWM output in Navio mode
INFO  [logger] logger started (mode=all)
INFO  [px4] Startup script returned successfully
INFO  [mavlink] using network interface wlan0, IP: 10.0.0.2
INFO  [mavlink] with netmask: 255.255.255.0
INFO  [mavlink] and broadcast IP: 10.0.0.255
INFO  [mavlink] partner IP: 10.0.0.24
WARN  [PreFlightCheck] Preflight Fail: Compass Sensor #0 missing
WARN  [PreFlightCheck] Preflight Fail: Accel Sensor #0 missing
WARN  [PreFlightCheck] Preflight Fail: Gyro Sensor #0 missing
WARN  [PreFlightCheck] Preflight Fail: Baro Sensor #0 missing

PX4 Exiting...
```

**Drone (please complete the following information):**
Quadcopter, DJI F450 airframe with DJI motors/ESCs
Navio2 on Ubuntu 20

blazczak, 2020-10-21:
More examples of the sensor timeouts in the log for #16008

mcsauder, 2020-10-21:
Hi @blazczak , does this issue and the other recent issues you have posted persist with the latest stable, v1.11.1, https://github.com/PX4/Firmware/releases/tag/v1.11.1, and also with current master?  Please post back if the issues exist with current master.

blazczak, 2020-10-21:
@mcsauder Yes, still there in v1.11.1 and master, both using emlid_navio2_default:

- v1.11.1
```
mpu9250 #0 on SPI bus 0 (devid=0x1)
lsm9ds1 #0 on SPI bus 0 (devid=0x3)
lsm9ds1_mag #0 on SPI bus 0 (devid=0x2)
ms5611 #0 on I2C bus 1 (external)
INFO  [px4] assuming working directory is rootfs, no symlinks needed.

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh posix-configs/rpi/px4.config 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
INFO  [mavlink] mode: Normal, data rate: 2880 B/s on /dev/ttyAMA0 @ 57600B
INFO  [logger] logger started (mode=all)
INFO  [px4] Startup script returned successfully
INFO  [mavlink] partner IP: 10.0.0.24
INFO  [ecl/EKF] 50095210: reset position to last known position
INFO  [ecl/EKF] 50095210: reset velocity to zero
INFO  [ecl/EKF] 56270093: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [mavlink] using network interface wlan0, IP: 10.0.0.2
INFO  [mavlink] with netmask: 255.255.255.0
INFO  [mavlink] and broadcast IP: 10.0.0.255
ERROR [sensors] Accel #0 fail:  TIMEOUT!
WARN  [ecl/EKF] baro hgt timeout - reset to baro
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [commander] Connection to ground station lost
INFO  [commander] Data link regained
WARN  [PreFlightCheck] Preflight Fail: Gyro 0 inconsistent - Check Cal
WARN  [PreFlightCheck] Preflight Fail: Yaw estimate error
WARN  [PreFlightCheck] Preflight Fail: High Gyro Bias
WARN  [ecl/EKF] baro hgt timeout - reset to gps
WARN  [ecl/EKF] baro hgt timeout - reset to baro
ERROR [sensors] Accel #1 fail:  TIMEOUT!

PX4 Exiting...
```

- latest master
```
mpu9250 #0 on SPI bus 0 (devid=0x1)
lsm9ds1 #0 on SPI bus 0 (devid=0x3)
lsm9ds1_mag #0 on SPI bus 0 (devid=0x2)
ms5611 #0 on I2C bus 1 (external)
INFO  [px4] assuming working directory is rootfs, no symlinks needed.

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh posix-configs/rpi/px4.config 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
INFO  [mavlink] mode: Normal, data rate: 2880 B/s on /dev/ttyAMA0 @ 57600B
[... 70 lines cut ...]
INFO  [commander] [cal] progress <61>
INFO  [commander] [cal] progress <81>
INFO  [commander] [cal] progress <100>
INFO  [commander] Updating SENS_BOARD_X_OFF -1.9 -> -5.9 degrees
INFO  [commander] Updating SENS_BOARD_Y_OFF 3.4 -> 3.1 degrees
INFO  [commander] [cal] calibration done: level
INFO  [commander] [cal] calibration started: 2 accel
INFO  [commander] [cal] pending: back front left right up down
INFO  [commander] [cal] hold vehicle still on a pending side
INFO  [commander] [cal] detected rest position, hold still...
INFO  [commander] [cal] down orientation detected
INFO  [commander] [cal] down orientation detected
INFO  [commander] [cal] Hold still, measuring down side
INFO  [commander] [cal] down side result: [-0.327 -0.993 -9.493]
INFO  [commander] [cal] progress <17>
INFO  [commander] [cal] down side done, rotate to a different side
INFO  [commander] [cal] down side done, rotate to a different side
INFO  [commander] [cal] pending: back front left right up
```

Master was actually worse than v1.11.1, sensor calibration failed twice, I'm assuming because of the issues related to the IMU devices we're discussing.

Based on what I've seen so far when building this for navio2 it looks like things might have gone sour after the transition from the IMU wrappers.

Given how per https://discuss.px4.io/t/navio2-px4-on-ubuntu/19007/12 it works fine for me in v1.10.1, I'll stick with that version for a while.

blazczak, 2020-11-05:
```
update: 1s, num topics: 66
TOPIC NAME                     INST #SUB RATE #Q SIZE
actuator_armed                    0    6    2  1   24 
actuator_controls_0               0    6  201  1   48 
adc_report                        0    1  100  1   96 
battery_status                    0    8  100  1  120 
commander_state                   0    1    2  1   16 
cpuload                           0    4    2  1   16 
estimator_innovation_test_ratios  0    1  102  1  152 
estimator_innovation_variances    0    1  102  1  152 
estimator_innovations             0    1  102  1  152 
estimator_sensor_bias             0    3  102  1  104 
estimator_states                  0    1  102  1  216 
estimator_status                  0    4  102  1  120 
rate_ctrl_status                  0    1  201  1   24 
sensor_accel                      0    4  398  8   48 
sensor_accel                      1    3  754  8   48 
sensor_baro                       0    2   73  1   32 
sensor_combined                   0    2  207  1   48 
sensor_gps                        0    4    5  1  104 
sensor_gyro                       0    4  398  8   40 
sensor_gyro                       1    3  754  8   40 
sensor_mag                        0    2   80  4   48 
update: 1s, num topics: 66
TOPIC NAME                     INST #SUB RATE #Q SIZE
actuator_armed                    0    6    2  1   24 
actuator_controls_0               0    6  202  1   48 
adc_report                        0    1  100  1   96 
battery_status                    0    8  100  1  120 
commander_state                   0    1    2  1   16 
cpuload                           0    4    2  1   16 
estimator_innovation_test_ratios  0    1  101  1  152 
estimator_innovation_variances    0    1  101  1  152 
estimator_innovations             0    1  101  1  152 
estimator_sensor_bias             0    3  101  1  104 
estimator_states                  0    1  101  1  216 
estimator_status                  0    4  101  1  120 
rate_ctrl_status                  0    1  202  1   24 
sensor_accel                      0    4  395  8   48 
sensor_accel                      1    3  750  8   48 
sensor_baro                       0    2   73  1   32 
sensor_combined                   0    2  209  1   48 
sensor_gps                        0    4    5  1  104 
sensor_gyro                       0    4  395  8   40 
sensor_gyro                       1    3  750  8   40 
sensor_mag                        0    2   79  4   48 
update: 1s, num topics: 66
TOPIC NAME                     INST #SUB RATE #Q SIZE
actuator_armed                    0    6    3  1   24 
actuator_controls_0               0    6  201  1   48 
adc_report                        0    1  100  1   96 
battery_status                    0    8  100  1  120 
commander_state                   0    1    3  1   16 
cpuload                           0    4    2  1   16 
estimator_innovation_test_ratios  0    1  100  1  152 
estimator_innovation_variances    0    1  100  1  152 
estimator_innovations             0    1  100  1  152 
estimator_sensor_bias             0    3  100  1  104 
estimator_states                  0    1  100  1  216 
estimator_status                  0    4  100  1  120 
led_control                       0    1    1  4   16 
rate_ctrl_status                  0    1  201  1   24 
sensor_accel                      0    4  397  8   48 
sensor_accel                      1    3  760  8   48 
sensor_baro                       0    2   73  1   32 
sensor_combined                   0    2  208  1   48 
sensor_gps                        0    4    5  1  104 
sensor_gyro                       0    4  397  8   40 
sensor_gyro                       1    3  760  8   40 
[... 401 lines cut ...]
top - 06:13:25 up 27 min,  1 user,  load average: 4.24, 2.53, 1.84
Threads: 199 total,   2 running, 197 sleeping,   0 stopped,   0 zombie
%Cpu(s):  7.8 us, 16.1 sy,  0.0 ni, 75.8 id,  0.0 wa,  0.0 hi,  0.3 si,  0.0 st
MiB Mem :    801.5 total,      5.1 free,    144.7 used,    651.7 buff/cache
MiB Swap:      0.0 total,      0.0 free,      0.0 used.    634.1 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND   
 2123 root      rt   0   62280   4172   3764 S  24.2   0.5   0:55.01 wq:SPI0   
 2096 root      rt   0   62280   4172   3764 S  14.8   0.5   0:23.30 wkr_hrt   
  649 root      20   0       0      0      0 D  10.0   0.0   2:35.85 rcio_work+
 2145 root      rt   0   62280   4172   3764 S   8.1   0.5   0:14.57 wq:manager
  149 root      20   0       0      0      0 S   6.5   0.0   1:34.41 spi0      
 2169 root     -51   0   62280   4172   3764 S   5.5   0.5   0:08.38 mavlink_i+
    1 root      20   0   33468   7868   5408 S   4.5   1.0   0:40.79 systemd   
 2140 root     -50   0   62280   4172   3764 D   4.2   0.5   0:06.60 gps       
 2135 root      rt   0   62280   4172   3764 S   3.9   0.5   0:06.07 wq:hp_def+
 2146 root      rt   0   62280   4172   3764 S   3.2   0.5   0:03.98 wq:rate_c+
  188 root      20   0       0      0      0 S   2.3   0.0   0:37.44 spi1      
 1120 root      20   0   12360   5252   4592 S   2.3   0.6   0:15.26 systemd-l+
 2173 root     -51   0   62280   4172   3764 S   2.3   0.5   0:03.62 mavlink_i+
 2130 root      rt   0   62280   4172   3764 S   1.6   0.5   0:02.57 wq:I2C1   
 2149 root     -91   0   62280   4172   3764 S   1.3   0.5   0:02.13 commander 
 2164 root      rt   0   62280   4172   3764 S   1.3   0.5   0:02.11 wq:manager
 2170 root     -51   0   62280   4172   3764 S   1.0   0.5   0:00.92 mavlink_r+
```

blazczak, 2020-11-09:
Had an opportunity to check against the latest master (defa437), still occurs.

dagar, 2020-11-09:
How does it change if you only run the mpu9250 (temporarily disable the lsm9ds1)?