PX4/PX4-Autopilot #16007: px4_raspberrypi 1.11.0 with Navio2 - QGroundControl display issues

blazczak, 2020-10-21:
**Describe the bug**
The battery indicator incorrectly shows 100%
The RC indicator in QGC does not light up

**To Reproduce**
Steps to reproduce the behavior:
Build 1.11.0 PX4 on Ubuntu 20, use instructions in https://community.emlid.com/t/navio2-support-for-ubuntu-20-04/19956 to create the spidev stuff if needed.

**Expected behavior**
Should see the correct battery level and RC connection indicators like in 1.10.1 on same hardware

**Drone**
Quadcopter, DJI F450 airframe with DJI motors/ESCs
Navio2 on Ubuntu 20

moreba1, 2020-10-23:
you should set bat1_v_div

blazczak, 2020-10-29:
@moreba1 I don't think what you're suggesting is the fix - this worked beautifully in 1.10.1 with the exact same settings; in both cases BAT_V_DIV is set (deprecated but the versioning code should take care of it).

moreba1, 2020-10-29:
yes, versioning code should take care of changing parameters name.
setting BAT1_ parameters is a faster solution.