PX4/PX4-Autopilot #16008: px4_raspberrypi 1.11.0 with Navio2 - trace level info in the logs and QGroundControl warning flashes

blazczak, 2020-10-21:
**Describe the bug**
A lot of small (debug) crud in the log and QGC warning flashes

**To Reproduce**
Steps to reproduce the behavior:
Build 1.11.0 PX4 on Ubuntu 20, use instructions in https://community.emlid.com/t/navio2-support-for-ubuntu-20-04/19956 to create the spidev stuff if needed.
Perform activity with the drone, like calibration, arming, etc.
See lots of extraneous messages in the logs and in the QGC warning flashes (the more annoying part) - possibly in-flight debug logging that was not cleaned up.

**Expected behavior**
Level of log sanitation seen in 1.10.1 for navio2 and fmu-v5

**Log Files and Screenshots**
```
mpu9250 #0 on SPI bus 0 (devid=0x1)
lsm9ds1 #0 on SPI bus 0 (devid=0x3)
lsm9ds1_mag #0 on SPI bus 0 (devid=0x2)
ms5611 #0 on I2C bus 1 (external, equal to '-b 1')
INFO  [px4] assuming working directory is rootfs, no symlinks needed.

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh posix-configs/rpi/px4.config 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
INFO  [linux_pwm_out] Loaded mixer from file ROMFS/px4fmu_common/mixers/quad_x.main.mix
INFO  [linux_pwm_out] Starting PWM output in Navio mode
INFO  [logger] logger started (mode=all)
INFO  [px4] Startup script returned successfully
INFO  [mavlink] using network interface wlan0, IP: 10.0.0.2
INFO  [mavlink] with netmask: 255.255.255.0
INFO  [mavlink] and broadcast IP: 10.0.0.255
INFO  [ekf2] Mag sensor ID changed to 4522498
INFO  [mavlink] partner IP: 10.0.0.24
INFO  [ecl/EKF] 1190242003: reset position to last known position
INFO  [ecl/EKF] 1190242003: reset velocity to zero
INFO  [ecl/EKF] 1197222218: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [commander] [cal] calibration started: 2 accel
INFO  [commander] [cal] pending: back front left right up down
INFO  [commander] [cal] hold vehicle still on a pending side
INFO  [commander] [cal] detected rest position, hold still...
INFO  [commander] [cal] down orientation detected
INFO  [commander] [cal] down orientation detected
INFO  [commander] [cal] Hold still, measuring down side
INFO  [commander] [cal] down side result: [  0.1153  -0.0607  -9.5143]
INFO  [commander] [cal] progress <17>
INFO  [commander] [cal] down side done, rotate to a different side
INFO  [commander] [cal] down side done, rotate to a different side
INFO  [commander] [cal] pending: back front left right up
INFO  [commander] [cal] hold vehicle still on a pending side
INFO  [commander] [cal] detected rest position, hold still...
INFO  [commander] [cal] detected motion, hold still...
INFO  [commander] [cal] detected rest position, hold still...
INFO  [commander] [cal] left orientation detected
INFO  [commander] [cal] left orientation detected
INFO  [commander] [cal] Hold still, measuring left side
INFO  [commander] [cal] left side result: [ -0.7990   9.7712   0.5110]
INFO  [commander] [cal] progress <34>
INFO  [commander] [cal] left side done, rotate to a different side
INFO  [commander] [cal] left side done, rotate to a different side
INFO  [commander] [cal] pending: back front right up
INFO  [commander] [cal] hold vehicle still on a pending side
[... 171 lines cut ...]
WARN  [commander] REJECT POSITION CONTROL
ERROR [sensors] Accel #0 fail:  TIMEOUT!
ERROR [sensors] Accel #0 fail:  TIMEOUT!
WARN  [commander] Connection to ground station lost
WARN  [PreFlightCheck] Preflight: GPS Horizontal Pos Error too high
INFO  [commander] Data link regained
WARN  [PreFlightCheck] Preflight Fail: no valid data from Accel #1
WARN  [PreFlightCheck] Preflight Fail: no valid data from Gyro #1
WARN  [PreFlightCheck] Preflight: GPS Horizontal Pos Error too high
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2020-10-20/04_42_29.ulg
INFO  [logger] Opened full log file: ./log/2020-10-20/04_42_29.ulg
INFO  [logger] Start mavlink log
INFO  [commander] Disarmed by auto preflight disarming
ERROR [logger] Ack timeout. Stopping mavlink log
INFO  [logger] closed logfile, bytes written: 180414
INFO  [logger] Stop mavlink log

PX4 Exiting...
Exiting NOW.
```

**Drone**
Quadcopter, DJI F450 airframe with DJI motors/ESCs
Navio2 on Ubuntu 20

dagar, 2020-10-21:
If you just let it run without calibration do the accels continue working with no timeout? You can check with `listener sensor_accel` or `sensors status`. What's the cpu load like with px4 running?

blazczak, 2020-10-21:
Still present in v1.11.1 and master.

@dagar Yes, I just let it sit there and got "ERROR [sensors] Accel #0 fail:  TIMEOUT!" out of the blue. Restarted, got the same error again.

In terms of cpu load, it's comparable to the load v1.10.1 generates and that deployment works and flies. It's the same clean Ubuntu 20 install, same uSD card, v1.10.1 and v1.11.1 checked out side-by-side, nothing else running on that instance - no offboard processes anything like that that would cause delays in sensor processing.

dagar, 2020-10-21:
I've only ever used the navio2 will the raspbian build from Emlid. With Ubuntu 20.04 are you using PREEMPT_RT? Are you running px4 as root?

blazczak, 2020-10-22:
When either 1.10.1 or 1.11.1 was selected it was run as root. No PREEMPT_RT, no ROS, just good ol' arm Ubuntu 20.04 LTS from https://ubuntu.com/download/raspberry-pi.

PX4 1.10.1 flies lively and issue-free in the same environment and on the same hardware: https://logs.px4.io/plot_app?log=4523b1b6-a0eb-41f1-9216-5c23d129a96b

dagar, 2020-10-22:
One notable difference for Navio2 since v1.11 is the switch to the unified drivers that are shared across Nuttx, Linux, Qurt. This applies to the mpu9250, lsm9ds1, ms5611, etc. In the case of the IMU drivers they actually transfer significantly more data now using the FIFOs (8 kHz vs 1 kHz). I'm wondering if that combined with no real time threads could be the problem.

What do you see in `uorb top`? While px4 is running can you share the output of `top` or `htop` (set to show all threads within px4)?
Then try stopping one of the `lsm9ds1` or `mpu9250`.

blazczak, 2020-11-05:
Posted in #16006