PX4/PX4-Autopilot #16009: make posix_sitl_default failed -Wfatal-errors

badsector998, 2020-10-21:
i have tried to follow tutorial on how using mavros in this link : https://gaas.gitbook.io/guide/software-realization-build-your-own-autonomous-drone/build-your-own-autonomous-drone-e01-offboard-control-and-gazebo-simulation

however, when i was trying to run "make posix_sitl_default" inside firmware directory it returns bunch of errors which i dont understand.

`ninja: Entering directory `/home/vergil/mavros_projects/1/catkin_ws/src/Firmware/build/posix_sitl_default'
[6/325] Building CXX object src/driver...Files/drivers__tfmini.dir/tfmini.cpp.o
FAILED: src/drivers/distance_sensor/tfmini/CMakeFiles/drivers__tfmini.dir/tfmini.cpp.o 
/usr/bin/ccache /usr/bin/c++  -DCONFIG_ARCH_BOARD_SITL -DMODULE_NAME=\"tfmini\" -DPX4_MAIN=tfmini_app_main -D__DF_LINUX -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/drivers/boards/sitl -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -I../../platforms/posix/include -Iexternal/Install/include -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wno-overloaded-virtual -Wreorder -Wno-format-truncation -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fdiagnostics-color=always -O2 -g -DNDEBUG   -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fdiagnostics-color=always -Wno-sign-compare -std=gnu++11 -MD -MT src/drivers/distance_sensor/tfmini/CMakeFiles/drivers__tfmini.dir/tfmini.cpp.o -MF src/drivers/distance_sensor/tfmini/CMakeFiles/drivers__tfmini.dir/tfmini.cpp.o.d -o src/drivers/distance_sensor/tfmini/CMakeFiles/drivers__tfmini.dir/tfmini.cpp.o -c ../../src/drivers/distance_sensor/tfmini/tfmini.cpp
In file included from /usr/include/string.h:495,
                 from ../../src/drivers/distance_sensor/tfmini/tfmini.cpp:53:
In function ‘char* strncpy(char*, const char*, size_t)’,
    inlined from ‘TFMINI::TFMINI(const char*, uint8_t)’ at ../../src/drivers/distance_sensor/tfmini/tfmini.cpp:193:9:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:34: error: ‘char* __builtin_strncpy(char*, const char*, long unsigned int)’ specified bound 20 equals destination size [-Werror=stringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[7/325] Building CXX object src/driver...MakeFiles/drivers__sf0x.dir/sf0x.cpp.o
FAILED: src/drivers/distance_sensor/sf0x/CMakeFiles/drivers__sf0x.dir/sf0x.cpp.o 
/usr/bin/ccache /usr/bin/c++  -DCONFIG_ARCH_BOARD_SITL -DMODULE_NAME=\"sf0x\" -DPX4_MAIN=sf0x_app_main -D__DF_LINUX -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/drivers/boards/sitl -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -I../../platforms/posix/include -Iexternal/Install/include -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wno-overloaded-virtual -Wreorder -Wno-format-truncation -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fdiagnostics-color=always -O2 -g -DNDEBUG   -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fdiagnostics-color=always -Wno-sign-compare -std=gnu++11 -MD -MT src/drivers/distance_sensor/sf0x/CMakeFiles/drivers__sf0x.dir/sf0x.cpp.o -MF src/drivers/distance_sensor/sf0x/CMakeFiles/drivers__sf0x.dir/sf0x.cpp.o.d -o src/drivers/distance_sensor/sf0x/CMakeFiles/drivers__sf0x.dir/sf0x.cpp.o -c ../../src/drivers/distance_sensor/sf0x/sf0x.cpp
In file included from /usr/include/string.h:495,
                 from ../../src/drivers/distance_sensor/sf0x/sf0x.cpp:51:
In function ‘char* strncpy(char*, const char*, size_t)’,
    inlined from ‘SF0X::SF0X(const char*, uint8_t)’ at ../../src/drivers/distance_sensor/sf0x/sf0x.cpp:196:9:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:34: error: ‘char* __builtin_strncpy(char*, const char*, long unsigned int)’ specified bound 20 equals destination size [-Werror=stringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[9/325] Building CXX object src/driver.../CMakeFiles/drivers__gps.dir/gps.cpp.o
FAILED: src/drivers/gps/CMakeFiles/drivers__gps.dir/gps.cpp.o 
/usr/bin/ccache /usr/bin/c++  -DCONFIG_ARCH_BOARD_SITL -DMODULE_NAME=\"gps\" -DPX4_MAIN=gps_app_main -D__DF_LINUX -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/drivers/boards/sitl -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -I../../platforms/posix/include -Iexternal/Install/include -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wno-overloaded-virtual -Wreorder -Wno-format-truncation -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fdiagnostics-color=always -O2 -g -DNDEBUG   -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fdiagnostics-color=always -std=gnu++11 -MD -MT src/drivers/gps/CMakeFiles/drivers__gps.dir/gps.cpp.o -MF src/drivers/gps/CMakeFiles/drivers__gps.dir/gps.cpp.o.d -o src/drivers/gps/CMakeFiles/drivers__gps.dir/gps.cpp.o -c ../../src/drivers/gps/gps.cpp
In file included from /usr/include/string.h:495,
                 from ../../src/drivers/gps/gps.cpp:59:
In function ‘char* strncpy(char*, const char*, size_t)’,
    inlined from ‘GPS::GPS(const char*, gps_driver_mode_t, GPSHelper::Interface, bool, bool, GPS::Instance)’ at ../../src/drivers/gps/gps.cpp:288:9:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:34: error: ‘char* __builtin_strncpy(char*, const char*, long unsigned int)’ specified bound 20 equals destination size [-Werror=stringop-truncation]
[... cut ...]
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:34: error: ‘char* __builtin_strncpy(char*, const char*, long unsigned int)’ specified bound 20 equals destination size [-Werror=stringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[15/324] Building CXX object src/syste...iles/systemcmds__param.dir/param.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:148: posix_sitl_default] Error 1
`

what is wrong??

dagar, 2020-10-22:
This appears to be quite an old version of PX4? Can you make sure you're on current master, I believe most of these string operation warnings were fixed.

Jaeyoung-Lim, 2020-10-22:
@badsector998 I think if you are using latest master, the correct command would be
```
make px4_sitl_default
```
or
```
make px4_sitl <simulator>
```
which <simulator> can be either `gazebo`, `jmavsim`, `jsbsim` depending on what you want to run

badsector998, 2020-10-22:
@dagar @Jaeyoung-Lim thanks for the reply. I'll try it again tomorrow in my workplace. I'll post again for the update 😃

badsector998, 2020-10-23:
@dagar @Jaeyoung-Lim here's the update.. i actually already have the newest version of this firmware. But it seems really consume a lot of resources to just do the make command and wrap it to gazebo. Is it really resources consuming? Even my laptop freezes a lot when do the make command.. and i haven't completed the make yet, i canceled it due to the freezing.

Jaeyoung-Lim, 2020-10-23:
What command were you using?

badsector998, 2020-10-23:
> What command were you using?

Just like you said
```
make px4_sitl gazebo
```