PX4/PX4-Autopilot #16022: implement new mag type for ekf2

moreba1, 2020-10-21:
In some conditions such as an indoor fly or fly in high magnetic fields that we will disable the mag fusion and select ekf2_mag_type to 5 (none), but the heading is wrong at the start of the flight. 

we can solve this problem by adding a new mag type that will use compass heading for setting heading when the battery connected and mag fusion will be disabled during flight.

https://github.com/PX4/Firmware/issues/11847

bresch, 2020-10-22:
For indoor flight, there is already a mode called "MC curstom" : 
*If set to 'MC custom' the behaviour is the same as 'Automatic, but if there are no earth frame position or velocity observations being used, the magnetometer will not be used. This enables vehicles to operate with no GPS in environments where the magnetic field cannot be used to provide a heading reference. Prior to flight, the yaw angle is assumed to be constant if movement tests controlled by the EKF2_MOVE_TEST parameter indicate that the vehicle is static. This allows the vehicle to be placed on the ground to learn the yaw gyro bias prior to flight.*

And flying in high magnetic fields but not before takeoff is quite unusual. For such use case (good GPS but bad mag), using a dual antenna GPS receiver (providing yaw) is recommended.

However, I think what we could anyway do is to initialize yaw with the mag (if available) even in `MAG_TYPE` "none"; that won't harm and could allow to takeoff in position/auto mode (before the yaw estimator converges) if the mag data before takeoff is good enough.

moreba1, 2020-10-22:
https://github.com/PX4/ecl/pull/918

https://github.com/PX4/ecl/pull/919

closed 2026-04-10 by github-actions[bot]