PX4/PX4-Autopilot #16028: pixhawk High Accelerometer bias warning

clqground, 2020-10-22:
How to solve the problem of "high accelerometer bias"?
pixhawk2.4.8
"high accelerometer bias" is often a problem when I arm the vehicle
After recalibrating the accelerometer , the vehicle is OK.
how  to solve the problem of "high accelerometer bias"
Does anyone have a solution?
Thank you very much for helping me.

dagar, 2020-10-22:
Which version of PX4 are you using? Can you reproduce this on v1.11.1 (the current stable)? There were improvements to the accel bias estimation when landed so that it's only done on the axis that's aligned with gravity (bias is observable).

clqground, 2020-10-24:
> Which version of PX4 are you using? Can you reproduce this on v1.11.1 (the current stable)? There were improvements to the accel bias estimation when landed so that it's only done on the axis that's aligned with gravity (bias is observable).

pixhawk2.4.8  firmware version V1.11.1 stable 
"high accelerometer bias " warning 
The accelerometer needs to be recalibrated every time the vehicle can work properly

clqground, 2020-10-24:
> Which version of PX4 are you using? Can you reproduce this on v1.11.1 (the current stable)? There were improvements to the accel bias estimation when landed so that it's only done on the axis that's aligned with gravity (bias is observable).

Thank you for your answer

bresch, 2020-10-26:
@clqground You might want to monitor the bias values after and before calibration `CAL_ACC._.OFF`. If they change, you might have issues with your sensor.

dagar, 2020-10-26:
@clqground it might help if you capture logs from boot (SDLOG_MODE) showing the problem, then the calibration (attempting to fix the accel bias), then the following log after a reboot.

clqground, 2020-10-27:
@dagar @bresch Thank you. I increased the value of the "COM_ARM_EKF_AB" parameter and now there is no problem
But this change may not be good
I will try your suggestions
thank you all