PX4/PX4-Autopilot #16035: v1.11 possible mag issues (Here2)

dagar, 2020-10-22:
- https://logs.px4.io/plot_app?log=e8f16d70-23a8-448d-8f2a-a9fa06a40892

mattpaddock, 2020-11-02:
I'll be flying a test frame on 1.11.1 w/ Here2 + Cube later this week. Will let you know what my results are.

mattpaddock, 2020-11-05:
@dagar Just tested this on a frame w/ Here2 + Cube. While it looked fine on the ground and was thoroughly calibrated, after taking off to 20 feet, it spit out a "Critical: Stopping compass use" error and then flew off in the wrong direction and crashed into the ground. 

Luckily it's a pretty rugged drone!

v1.10 on the same drone has no issues.

mattpaddock, 2020-11-09:
@mrpollo can't share raw flight logs due to govt customer policy. If there's a specific value you could use the log of, I'd be more than happy to graph it out and send a screenshot.

ryanjAA, 2020-11-10:
@mattpaddock what if the gps data is scrubbed or shifted, etc. we had this (log anonymity need) issue awhile back and iirc @dagar had some ideas for solutions so the logs could be uploaded and used for analysis.

ryanjAA, 2020-11-10:
@mattpaddock was this on a multirotor or fixed-wing you had the mag issue on?

mattpaddock, 2020-11-10:
@ryanjAA a feature like that would certainly help us in sharing more logs. We're legally limited on what we can share fin the form of logs, software, etc.

The issue was on a multirotor (PX4 Generic Hex frame).

RomanBapst, 2020-11-10:
@mattpaddock This is actually quite interesting. Could you share the graphs for the "yaw_estimator_status" topic?
All 5 yaw values would be nice to see.

mattpaddock, 2020-11-17:
> @mattpaddock This is actually quite interesting. Could you share the graphs for the "yaw_estimator_status" topic?
> All 5 yaw values would be nice to see.

I'm actually under COVID-quarantine right now (had to go out of state for a customer demonstration) but I'm expecting a negative test result that'll get me back in the office by late this week.

Next time I'm in I'll pull the logs and send the graphs!

mattpaddock, 2020-11-25:
@RomanBapst Screenshots below show the yaw estimator status topics in FlightPlot. If you want me to focus on one, let me know.

I've also included a picture of the log messages that I thought was interesting, particularly the "no global position". I repeated this flight twice and had the same results on both flights. Downgrading to 1.10 immediately resolved all issues.

[image]
[image]

ryanjAA, 2020-11-25:
@mattpaddock A try with 1.11.2 would be worthwhile. That has https://github.com/PX4/PX4-ECL/pull/914 in it.