PX4/PX4-Autopilot #16057: VTOL Tailsitter: Thrust issue during transition

RicardoM17, 2020-10-27:
**Describe the bug**
When we are performing a VTOL transition with a tailsitter type drone We are getting very weird thrust fluctuations. It seems that it's taking two references basically at the same time, I would assume from the MC and FW controllers.
Also in simulation we don't get this issue.

More over @xdwgood also reported this issue on PX4 slack, thus proving that this issue is on upstream PX4 and not custom software.

**To Reproduce**
Use a VTOL Tailsitter type drone and perform a transition.

**Expected behavior**
The thrust reference should be smooth during transition.

**Log Files and Screenshots**

[image]

https://logs.px4.io/plot_app?log=1e0b41e7-79d0-4d75-be15-27168f4274cc

[image]

https://review.px4.io/plot_app?log=6ce3fbeb-33d0-44db-8232-f3f9dd0f87b2

**Additional context**
@xdwgood reported that setting MPC_USE_HTE to 0, i.e. disabling the hover throttle estimator, seems to solve the issue. It's still to be confirmed but I'm still creating this issue to keep track of the problem as it should be fixed entirely.

@sfuhrer @RomanBapst  FYI

xdwgood, 2020-10-27:
@dagar  I think the problem is caused by hovering thrust estimation. here :

```
		if (_vehicle_local_position_setpoint_sub.update(&local_pos_sp)) {
			if (PX4_ISFINITE(local_pos_sp.thrust[2])) {
				// Inform the hover thrust estimator about the measured vertical
				// acceleration (positive acceleration is up) and the current thrust (positive thrust is up)
				ZeroOrderHoverThrustEkf::status status;
				_hover_thrust_ekf.fuseAccZ(-local_pos.az, -local_pos_sp.thrust[2], status);

				const bool valid = _in_air && (status.hover_thrust_var < 0.001f) && (status.innov_test_ratio < 1.f);
				_valid_hysteresis.set_state_and_update(valid, local_pos.timestamp);
				_valid = _valid_hysteresis.get_state();

				publishStatus(local_pos.timestamp, status);
			}
		}

	} else {
		_valid_hysteresis.set_state_and_update(false, hrt_absolute_time());

		if (!_armed) {
			reset();
		}

		if (_valid) {
			// only publish a single message to invalidate
			publishInvalidStatus();

			_valid = false;
		}
	}
```

Even if valid is 0, valid information is still released.
[image]

RicardoM17, 2020-10-27:
I would also like to add that in my case it seems to be related to MPC_THR_MIN. changing this value seems to change the value of the "clipping" that is happening with the thrust during transition.

It could still be related to the proposed cause above however.

xdwgood, 2020-10-27:
> I would also like to add that in my case it seems to be related to MPC_THR_MIN. changing this value seems to change the value of the "clipping" that is happening with the thrust during transition.

It's the same situation for me

bresch, 2020-10-27:
@RicardoM17 In your log, the solution of HTE is never declared valid, so I don't think the value itself is doing anything
[image]
I'll check if setting `MPC_USE_HTE` to 1 makes other undesired effects, even if the estimate is never valid

bresch, 2020-10-27:
To me, it actually looks that the issue is in the "blending logic" of MC and FW as it seems to toggle between two values
[image]

bresch, 2020-10-27:
The two logs don't seem to have the same firmware version (the 1st one doesn't have the validity flag of HTE) and the issue actually looks different to me. Below is a screenshot of the 1st log, and above was the 2nd one:
[image]

yuthikasagarage, 2020-10-27:
I'm building on top of 1.10,  i haven't had this issue in my test flights. 
i need to use more powerful motors, i played with the one i had.
https://review.px4.io/plot_app?log=071608a6-c25e-4887-90ec-d860e86a6382

xdwgood, 2020-10-28:
@yuthikasagarage  Your log does not explain the problem, your log shows that the hovering throttle has not worked, so you are normal

xdwgood, 2020-10-28:
@bresch  Thank you for your reply, the second log has made minor changes based on this commit　　　
0b0d23e8fa636da1a06ad24fee1bb73eb5c737aa

Regarding the second log, I also noticed that the effective value of the thrust estimate is 0, but I think this is weird. Why does the thrust estimate start to release some values only during transmission? I think this is abnormal. So far I have not found direct evidence from the code that the problem lies in hovering thrust estimation. I will continue to investigate.

> 　I would also like to add that in my case it seems to be related to MPC_THR_MIN. changing this value seems to change the value of the "clipping" that is happening with the thrust during transition.

It's the same situation for me. Based on this, I suspect that we are the same problem

xdwgood, 2020-10-28:
After disabling the use of the airspeed sensor, I found this problem again in the simulation

https://review.px4.io/plot_app?log=0278e9c4-5329-4b8c-8d1c-7d3e8c5cee90

xdwgood, 2020-10-28:
This is a normal log (in flight simulation)

[image]

This is a flight log with errors

[image]

Note: vehicle_attitude_setpoint_0.thrust_body[2]  x  -1

Under normal circumstances, these three values should be equal, so the problem may occur in local.thrust[2]

bresch, 2020-10-28:
> Under normal circumstances, these three values should be equal, so the problem may occur in local.thrust[2]

Good catch. Is it maybe related due to the tailsitter doing its rotation by 90 degrees?

RicardoM17, 2020-10-28:
> To me, it actually looks that the issue is in the "blending logic" of MC and FW as it seems to toggle between two values

This was my original guess.

> Good catch. Is it maybe related due to the tailsitter doing its rotation by 90 degrees?

Could be. I don't think it would also be the first time :sweat_smile:

xdwgood, 2020-10-29:
Now I almost locked the problem. This problem only occurs when converting in manual/stabilized mode. This is because there are two places inside the program that publish the same information.

https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/mc_att_control/mc_att_control_main.cpp#L213-L216 
https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/mc_pos_control/MulticopterPositionControl.cpp#L443

In my opinion, pilot control should be followed in manual mode, even during transmission. @sfuhrer  @bresch  Any suggestions?

sfuhrer, 2020-10-29:
> In my opinion, pilot control should be followed in manual mode, even during transmission. @sfuhrer @bresch Any suggestions?
Could well be. Can you check how it looks like with this commit cherry-picked: https://github.com/PX4/PX4-Autopilot/commit/99dbdb73b64a37af2327a346c6cea24d8f204f26 . It disables the transition flight task for tailsitters in manual mode.

xdwgood, 2020-10-29:
```
			// publish attitude setpoint
			// It's important to publish also when disarmed otheriwse the attitude setpoint stays uninitialized.
			// Not publishing when not running a flight task
			// in stabilized mode attitude setpoints get ignored
			// in offboard with attitude setpoints they come from MAVLink directly

			if ((_vehicle_status.nav_state == vehicle_status_s::NAVIGATION_STATE_MANUAL ||
			_vehicle_status.nav_state == vehicle_status_s::NAVIGATION_STATE_STAB) &&
			_vehicle_status.in_transition_mode) {

			} else {
				_vehicle_attitude_setpoint_pub.publish(attitude_setpoint);
			}
```

I added the above code, below is the log before and after the change (simulation flight test)

before:
[image]

after:
[image]

xdwgood, 2020-10-29:
@sfuhrer  For me, just adding this `_control_mode.flag_control_altitude_enabled` condition is enough, and add some necessary notes

sfuhrer, 2021-01-18:
@RicardoM17 this should be fixed with https://github.com/PX4/PX4-Autopilot/pull/16244 right?

closed 2021-01-18 by sfuhrer

RicardoM17, 2021-01-19:
> @RicardoM17 this should be fixed with #16244 right? Please re-open if not.

This is correct.

Thanks for closing the issue.