PX4/PX4-Autopilot #16070: Compass calibration resets

LaurensKools, 2020-10-28:
Hi everyone!

I’m using a Cube Orange with PX4 1.11.0, Here 2 GPS and Qgroundcontrol V4.0. Every time the system reboots (either via software or unplug the power), the compass calibration resets and the heading is inconsistent (pre-flight fail: Compasses inconsistent by x degrees). When recalibrating all sensors (gyro, accel, level horizon and compass) and not click “reboot” it does work fine. However, doing this every time after a reboot shouldn’t be necessary so is there a solution to this problem?

dagar, 2020-10-29:
Are you losing other parameters on reboot?

LaurensKools, 2020-10-31:
No that's not the case. I also tried different orange cube's but same issue appears

LaurensKools, 2020-11-03:
@dagar do you have an idea how to solve this?

dagar, 2020-11-03:
> @dagar do you have an idea how to solve this?

Could you verify exactly what's on your system?

From the command line (serial console or through QGC mavlink console).

`work_queue status`
`listener sensor_mag`
`param show CAL_MAG*`

LaurensKools, 2020-11-03:
@dagar yes sure, captured the results below:
[image]

dagar, 2020-11-03:
Ok, so it looks like the Here2 isn't being detected at all.

julianoes, 2023-04-03:
Closing this, follow up in https://github.com/PX4/PX4-Autopilot/issues/15591

closed 2023-04-03 by julianoes