PX4/PX4-Autopilot #16073: SET_POSITION_TARGET_GLOBAL_INT does not reject invalid input.

mattpaddock, 2020-10-28:
**Describe the bug**
When using SET_POSITION_TARGET_GLOBAL_INT in offboard mode, it accepts invalid input while still providing "warnings" that the input is invalid. When using SET_POSITION_TARGET_GLOBAL_INT with invalid input, like MAV_FRAME_GLOBAL_RELATIVE_ALT, the drone will start to go to the position target while warning, and eventually time out and crash.

**To Reproduce**
Steps to reproduce the behavior:
1. Takeoff drone.
2. Set drone into offboard mode.
3. Send SET_POSITION_TARGET_GLOBAL_INT with MAV_FRAME_GLOBAL_RELATIVE_ALT at 5-10hz.
4. See error

**Expected behavior**
Vehicle should reject the command entirely and send a warning.

**Drone (please complete the following information):**
- PX4 Quadcopter (v1.11.1)

Jaeyoung-Lim, 2020-12-06:
@mattpaddock Would you be able to provide a log for the crash?

Jaeyoung-Lim, 2021-01-08:
@mattpaddock Your problem should be fixed in https://github.com/PX4/PX4-Autopilot/pull/16514, but this does not check if the input is invalid.

Offboard mode in general is only for development / prototyping, therefore there it is the intention to disable safety checks to support this usecase. Therefore, IMO the developer should be responsible for checking if the inputs are valid when using offboard mode.

closed 2021-01-08 by Jaeyoung-Lim