PX4/PX4-Autopilot #16075: Incorrect Python executable selected when configuring under VSCode

haplm, 2020-10-28:
**Describe the bug**

This is related to #14077. I'm on macOS (latest Catalina 10.15.7)

Building works perfectly fine on command line, it correctly finds my Python 3.9 installation in pyenv, where all the required packages are available.

However, when I try clean build under VS Code, the configuration phase finds system Python without the dependencies. However, I made sure that the the shell in VSCode is using the right Python, and even set the PYTHON_EXECUTABLE variable.

This is invoked in my VS Code terminal
```
haplm@Martins-MBP Firmware % which python
/Users/haplm/.pyenv/shims/python
haplm@Martins-MBP Firmware % /Users/haplm/.pyenv/shims/python -V  
Python 3.9.0
haplm@Martins-MBP Firmware % echo $PYTHON_EXECUTABLE 
/Users/haplm/.pyenv/shims/python
```

If I first build on command line, and then in VS Code, it works OK.

I'm using ZSH shell (which Apple now pushes to all machines), so I have altered ```.vscode/tasks/json``` this way:
```
    "tasks": [
        {
            "label": "build",
            "type": "shell",
            "command": "/bin/zsh",
```

Also my terminal in VS Code is switched to ZSH. 

I have even tried to configure bash to reflect my zsh setup, but that doesn't help either. So the VS Code scripts has to bo doing something wrong.

dagar, 2020-10-28:
Do you see the python version on the bottom blue bar?

[image]

haplm, 2020-10-28:
Yes I do. There's the correct one:

[image]

dagar, 2020-10-29:
Are you able to catch the cmake configure output in vscode from a clean state?

[image]

haplm, 2020-10-29:
Sure, sorry for not mentioning it earlier, this is where I see that it finds the wrong interpreter:
```
[variant] Loaded new set of variants
[kit] Successfully loaded 4 kits from /Users/haplm/.local/share/CMakeTools/cmake-tools-kits.json
[kit] Successfully loaded 1 kits from /Users/haplm/Development/px4/Firmware/.vscode/cmake-kits.json
[main] Configuring folder: Firmware 
[proc] Executing command: /usr/local/bin/cmake --no-warn-unused-cli -DCONFIG:STRING=holybro_kakutef7_default -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=MinSizeRel -H/Users/haplm/Development/px4/Firmware -B/Users/haplm/Development/px4/Firmware/build/holybro_kakutef7_default -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] -- PX4 version: v1.11.0-rc3-563-gb1dc1b1ecd
[cmake] -- PX4 config file: /Users/haplm/Development/px4/Firmware/boards/holybro/kakutef7/default.cmake
[cmake] -- PX4 config: holybro_kakutef7_default
[cmake] -- PX4 platform: nuttx
[cmake] -- cmake build type: MinSizeRel
[cmake] -- The CXX compiler identification is GNU 9.2.1
[cmake] -- The C compiler identification is GNU 9.2.1
[cmake] -- The ASM compiler identification is GNU
[cmake] -- Found assembler: /usr/local/bin/arm-none-eabi-gcc
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++ - skipped
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc - skipped
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- ccache enabled (export CCACHE_DISABLE=1 to disable)
[cmake] -- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.9", minimum required is "3") 
[cmake] -- build type is MinSizeRel
[cmake] -- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-459-gdd3ffc4
[cmake] -- ROMFS: px4fmu_common
[cmake] -- ROMFS: Adding rc.board_defaults
[cmake] -- ROMFS: Adding rc.board_sensors
[cmake] -- ROMFS: Adding rc.board_extras
[cmake] -- ROMFS: Adding rc.board_mavlink
[cmake] -- Found SVD: 
[cmake] -- Configuring done
[cmake] -- Generating done
[cmake] -- Build files have been written to: /Users/haplm/Development/px4/Firmware/build/holybro_kakutef7_default
[cmakefileapi-parser] Code model version (2.1) of cmake-file-api is unexpected. Expecting (2.0). IntelliSense configuration may be incorrect.
[cmakefileapi-parser] Code model version (2.1) of cmake-file-api is unexpected. Expecting (2.0). IntelliSense configuration may be incorrect.
```