PX4/PX4-Autopilot #16083: System crash

jiejiezhang, 2020-10-29:
platform: stm32f427(Pixhaw v3)

ISSUE:  System crashes always occur, and it is found that it may be related to serial cache

Anyone can help to see this? Thanks very much!

{"data":{"dsk":"D1648k1bu0rfi8","dss":"8fbdbaed6e4f1895","capabilities":[{"expireTime":1630425599999,"staup_hardfault: Hard Fault:
up_hardfault:   IRQ: 3 regs: 0x1000a534
up_hardfault:   BASEPRI: 000000f0 PRIMASK: 00000000 IPSR: 00000003 CONTROL: 00000000
up_hardfault:   CFAULTS: 00010000 HFAULTS: 40000000 DFAULTS: 00000000 BFAULTADDR: e000ed38 AFAULTS: 00000000
up_hardfault: PANIC!!! Hard fault: 40000000
up_assert: Assertion failed at file:armv7-m/up_hardfault.c line: 148 task: wq:UART3
up_registerdump: R0: 10004870 1000430b 080e5ab4 00000031 00000000 20006d2c 1000a850 1000a608
up_registerdump: R8: 00000000 00000000 00000000 00000000 00000000 1000a608 08085bc3 08084c5a
up_registerdump: xPSR: 81000000 BASEPRI: 000000f0 CONTROL: 00000000
up_registerdump: EXC_RETURN: ffffffe9
up_dumpstate: sp:     20002f98
up_dumpstate: IRQ stack:
up_dumpstate:   base: 20003000
up_dumpstate:   size: 00000200
up_dumpstate:   used: 00000158
up_stackdump: 20002f80: 00000000 1000a8c8 20002f98 20003000 0000270c 08005a21 000000f0 00000000
up_stackdump: 20002fa0: 00000000 00000000 1000a608 08085bc3 08084c5a 20002fc0 080057d1 00000003
up_stackdump: 20002fc0: 00000000 080057d9 40000000 00000000 e000ed38 00000000 20003000 0800735b
up_stackdump: 20002fe0: 000000f0 08005859 000000f0 1000a534 20006d2c 1000a850 1000a608 08004fbb
up_dumpstate: sp:     1000a608
up_dumpstate: User stack:
up_dumpstate:   base: 1000a8c8
up_dumpstate:   size: 0000270c
up_dumpstate:   used: 00000d28
up_stackdump: 1000a600: 00000010 00000000 1000430b 10004870 00000001 00000000 00000000 00000000
up_stackdump: 1000a620: 00000000 3ff00000 00000000 00000000 100042fe 00000000 00000000 08076071
up_stackdump: 1000a640: 1000a648 08085bc3 0000000a 2000938c 1000430b 10004870 1000a660 0808646b
up_stackdump: 1000a660: 10004840 2000938c 1000430a 10004840 20006d2c 10004870 1000a680 08085bef
up_stackdump: 1000a680: 00000000 2000938c 100042fd 10004840 1000a698 08085ffd 100070e0 2000938c
up_stackdump: 1000a6a0: 100042fd 100070e0 20006d2c 10004840 1000a6b8 08085bd9 0000000c 2000938c
up_stackdump: 1000a6c0: 100042fc 100070e0 1000a6d0 080864ff 1000aae0 2000938c 100042fb 1000aae0
up_stackdump: 1000a6e0: 100070e0 100070e0 1000a6f0 08085bef 00000004 2000938c 100042bc 1000aae0
up_stackdump: 1000a700: 1000a708 0808646b 10004380 2000938c 100042bb 10004380 20006d2c 1000aae0
up_stackdump: 1000a720: 1000a728 08085bef 00000000 2000938c 100042b4 10004380 1000a740 08085a25
up_stackdump: 1000a740: ffffffff 00000000 00000000 100042b4 1000a780 10004380 2000938c 00000000
up_stackdump: 1000a760: 1000a768 08085a8b 00000000 100042b4 1000a778 08076005 00000000 1000a7ac
up_stackdump: 1000a780: 030e4534 100042b4 00000000 00000000 1000a798 080760bb 100051e0 000000a8
up_stackdump: 1000a7a0: 100042b4 000000c8 1000a7b0 00000000 100051e0 00000000 1000a7c0 0807919d
up_stackdump: 1000a7c0: 386d4396 00000000 1000aa80 1000aa80 1000aa80 00000000 1000a7c8 1000aa80
up_stackdump: 1000a7e0: 10007114 10007808 1000a7f0 080729f3 386d4396 00000000 386d4396 00000000
up_stackdump: 1000a800: 1000a848 08071fa7 100077c0 100077c0 1000a850 08046193 1000a840 080a4161
up_stackdump: 1000a820: 00000000 1000a840 2000c298 10003ba0 10003184 080a41f5 10003058 10003ba4
up_stackdump: 1000a840: 1000b518 080087f0 100077c0 100077c0 00010001 00000000 00000000 00000000
up_stackdump: 1000a860: 00000000 080eb6c0 100077c0 00000000 00000001 00000000 00000000 00000000
up_stackdump: 1000a880: 00000000 0001ffff 00000000 00000000 00000000 00000000 00000000 10005200
up_stackdump: 1000a8a0: 100052e0 10004670 10003160 080188ab 00000000 00000000 00000000 00000000
up_stackdump: 1000a8c0: 00000000 00000000 deadbeef a9776bd0 5769e6e1 6adf093c 000001a0 00002720
up_taskdump: Idle Task: PID=0 Stack Used=0 of 0
up_taskdump: hpwork: PID=1 Stack Used=344 of 1260
up_taskdump: lpwork: PID=2 Stack Used=344 of 1612
up_taskdump: init: PID=3 Stack Used=1848 of 2924
up_taskdump: wq:manager: PID=4 Stack Used=384 of 1252
up_taskdump: mavlink_if0: PID=69 Stack Used=1328 of 2572
up_taskdump: gps: PID=79 Stack Used=960 of 1676
up_taskdump: wq:UART3: PID=17 Stack Used=3368 of 9996
up_taskdump: wq:navigation_and_contro: PID=52 Stack Used=1072 of 7196
up_taskdump: wq:lp_default: PID=21 Stack Used=780 of 1700
up_taskdump: wq:rate_ctrl: PID=54 Stack Used=704 of 1660
up_taskdump: wq:hp_default: PID=27 Stack Used=916 of 1900

dagar, 2020-10-29:
What do you have on `UART3`?

jiejiezhang, 2020-10-29:
Uart3 is a 4G module. It communicates with the background by using the Qianxun SDK and requests the RTCM data stream

closed 2020-10-30 by jiejiezhang