PX4/PX4-Autopilot #16095: No px4flow parameter SENS_EN_FLOW in intel_aerofc-v1_default.px4

DRAEYE, 2020-10-31:
Hi, I'm using the intel aero RTF with an additionally px4flow module, and using the latest intel_aerofc-v1_default.px4, but no param called SENS_EN_FLOW is found in QGC, the param is listed in doc, how I can enable the px4flow

dagar, 2020-10-31:
It's not included in the build by default (likely to save flash).

You can add it by commenting this line. https://github.com/PX4/PX4-Autopilot/blob/4df3931bd14c3de4e92fab26ede2b9d51e7cb948/boards/intel/aerofc-v1/default.cmake#L25

DRAEYE, 2020-11-01:
Thank you, I am trying that and successfully there is px4flow command in nsh, another question is if I connect the px4flow with the telemetry port (UART), where should I modify the baudrate in firmware (from 57600 to 115200) to recognize the px4flow

closed 2020-11-01 by DRAEYE

dagar, 2020-11-01:
> Thank you, I am trying that and successfully there is px4flow command in nsh, another question is if I connect the px4flow with the telemetry port (UART), where should I modify the baudrate in firmware (from 57600 to 115200) to recognize the px4flow

It's mavlink on the UART, and I believe the baudrate is 115200.