PX4/PX4-Autopilot #16098: Had the same issues and re-downloaded gazebo plug-ins

fearless-pilgrim, 2020-11-01:
Had the same issues and re-downloaded gazebo plug-ins
`sudo apt-get install libprotobuf-dev libprotoc-dev protobuf-compiler libeigen3-dev libxml2-utils python-rospkg python-jinja2`
`sudo apt-get install libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-bad gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly -y`

and it worked like a charm

_Originally posted by @alithethird in https://github.com/PX4/PX4-Autopilot/issues/13759#issuecomment-623235991_

Jaeyoung-Lim, 2020-11-02:
@fearless-pilgrim Thanks for sharing, Is there a reason why you re-posted this from the original issue?

Jaeyoung-Lim, 2021-01-17:
Closing as author didn't respond, and it is a duplicate

closed 2021-01-17 by Jaeyoung-Lim