PX4/PX4-Autopilot #16101: I cant build project for gazebo

andrey-pr, 2020-11-02:
When i try to build project for gazebo, my computer freezes for few minutes? and later i get this text:

$ make px4_sitl gazebo_rover
[0/4] Performing build step for 'sitl_gazebo'
[1/66] Building CXX object CMakeFile...ugin.dir/src/gazebo_gps_plugin.cpp.o
FAILED: CMakeFiles/gazebo_gps_plugin.dir/src/gazebo_gps_plugin.cpp.o 
/usr/bin/ccache /usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DLIBBULLET_VERSION=2.88 -DLIBBULLET_VERSION_GT_282 -Dgazebo_gps_plugin_EXPORTS -I/home/andrey/Documents/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-11/gazebo/msgs -I/home/andrey/Documents/Firmware/mavlink/include -I/home/andrey/Documents/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/gazebo-11 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-9.3 -isystem /usr/include/ignition/math6 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport8 -isystem /usr/include/ignition/msgs5 -isystem /usr/include/ignition/common3 -isystem /usr/include/ignition/fuel_tools4 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -isystem /usr/include/eigen3 -isystem /usr/include/opencv4 -isystem /usr/include/sdformat-9.3/sdf/.. -isystem /usr/include/ignition/cmake2 -isystem /usr/include/uuid -O2 -g -DNDEBUG -fPIC   -Wno-deprecated-declarations -Wno-address-of-packed-member -I/usr/include/uuid -std=gnu++17 -MD -MT CMakeFiles/gazebo_gps_plugin.dir/src/gazebo_gps_plugin.cpp.o -MF CMakeFiles/gazebo_gps_plugin.dir/src/gazebo_gps_plugin.cpp.o.d -o CMakeFiles/gazebo_gps_plugin.dir/src/gazebo_gps_plugin.cpp.o -c /home/andrey/Documents/Firmware/Tools/sitl_gazebo/src/gazebo_gps_plugin.cpp
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
[2/66] Building CXX object CMakeFile...r/src/gazebo_gst_camera_plugin.cpp.o
FAILED: CMakeFiles/gazebo_gst_camera_plugin.dir/src/gazebo_gst_camera_plugin.cpp.o 
/usr/bin/ccache /usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DLIBBULLET_VERSION=2.88 -DLIBBULLET_VERSION_GT_282 -Dgazebo_gst_camera_plugin_EXPORTS -I/home/andrey/Documents/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-11/gazebo/msgs -I/home/andrey/Documents/Firmware/mavlink/include -I/home/andrey/Documents/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/gazebo-11 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-9.3 -isystem /usr/include/ignition/math6 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport8 -isystem /usr/include/ignition/msgs5 -isystem /usr/include/ignition/common3 -isystem /usr/include/ignition/fuel_tools4 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -isystem /usr/include/eigen3 -isystem /usr/include/opencv4 -isystem /usr/include/sdformat-9.3/sdf/.. -isystem /usr/include/ignition/cmake2 -isystem /usr/include/uuid -O2 -g -DNDEBUG -fPIC   -Wno-deprecated-declarations -Wno-address-of-packed-member -I/usr/include/uuid -std=gnu++17 -MD -MT CMakeFiles/gazebo_gst_camera_plugin.dir/src/gazebo_gst_camera_plugin.cpp.o -MF CMakeFiles/gazebo_gst_camera_plugin.dir/src/gazebo_gst_camera_plugin.cpp.o.d -o CMakeFiles/gazebo_gst_camera_plugin.dir/src/gazebo_gst_camera_plugin.cpp.o -c /home/andrey/Documents/Firmware/Tools/sitl_gazebo/src/gazebo_gst_camera_plugin.cpp
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
^Cninja: build stopped: interrupted by user.
ninja: build stopped: interrupted by user.
make: *** [Makefile:226: px4_sitl] Interrupt

Jaeyoung-Lim, 2020-11-02:
@andrey-pr Maybe your system runs out of memory while compiling the project?

andrey-pr, 2020-11-02:
> @andrey-pr Maybe your system runs out of memory while compiling the project?

I have 8gb of memory. I think its enought. And building project for jmavsim working normally

andrey-pr, 2020-11-02:
> @andrey-pr Maybe your system runs out of memory while compiling the project?

Thanks. Indeed, it was a lack of RAM. I enabled 4gb swap and its started working

closed 2020-11-02 by andrey-pr