PX4/PX4-Autopilot #16110: Batt_smbus need to be restructured to handle different smart battery systems in a cleaner architecture

echoGee, 2020-11-03:
**Describe problem solved by the proposed feature**
There are various different smart battery systems all of which use an underlying SMBUS protocol, which is not comprehensive. batt_smbus module has be refactored to handle the different systems.

**Describe your preferred solution**
Ideal solution is that there is one overarching protocol that everyone uses. But this is utopia/unrealistic since there are already folks having their own implementation which differ from others. Eg bq40z80, solo battery, batmon etc

**Describe possible alternatives**
Build a base class with the minimal implementation of SMBUS. Inherit this class for building a product specific implementation which can be initiated. 
This is nicely done by ardupilot here: https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_BattMonitor 
This feature request is a placeholder to get the ball rolling with the maintainers/code owners.

echoGee, 2021-01-22:
Hi, 
Any suggestions on how to handle this ? 
From how i see it, the main steps to make this happen are :

1. Move all of batt_smbus functions that are supported in the SMBUS formal specs to a base class (batt_smbus)
2. Implement derived classes specific for bq40z80, bq40z50, solo, batmon
3. Have a function that initializes the right subclass based on a flag passed on from QGC. It seems GPS driver is doing something similar to this (https://github.com/PX4/PX4-Autopilot/blob/fa5a6cf712937a66cb07e3ba2682cf08347cad5b/src/drivers/gps/gps.cpp#L760) ? Is that the right approach? 
4. Should there be a default initialization of batt_smbus class thats SMBUS only that talks the basic SMBUS data only? 

@BazookaJoe1900 @dakejahl

BazookaJoe1900, 2021-01-23:
for the differentiation between the bq40z50 and bq40z80 only few if/else where needed. so the need for more complex hierarchy seems to me unnecessary.
I don't familiar the solo, and the batmon battery protocol. but my instincts tell me that if their protocol is similar to the bq, its better not design diverted class's. but to use some more if/else

echoGee, 2021-01-25:
Even if the differentiation between bq40zx0 and other BQ chips (BQ769x0) etc have only one difference, all the difference would have to be handled in an if else case. Pros and Cons of an inherited system:
Cons: 
- Someone has to do this work. 
- No immediate functionality difference
- Requires testing on all the current different hardware (BQ chips, solo etc ? )
- Some, but trivial duplication of code

Pros:
- Much cleaner code. Each device class handles its own functionality, nothing more
- Ability to handle chips in same series like BQ76940,BQ76930 for BQ769x0 in an efficient way with a series specific base class if there is a need for this. 
- Ability to handle chip specific settings (like seal unseal etc) which are not part of standard SMBUS spec , but a practical necessity in a chip specific driver. Each chip driver can be maintained/refined without need to test on other chips

dagar, 2021-01-25:
@echoGee My personal preference these days is to distill the driver down to the simplest possible form targeted at one exact model. What I saw with things like unified Invensense IMU drivers is that a lot of the minor differences get lost in the details and can come back to bite you later on. This creates a bit of bloat in generic PX4 builds for development boards (eg pixhawk 4), but it doesn't really matter for actual products where you can customize the build to only support include what you physically have.

I would only suggest a more unified approach if someone actually has a deep understanding of the full series so that a properly abstracted general driver can be written.

echoGee, 2021-01-25:
> @echoGee My personal preference these days is to distill the driver down to the simplest possible form targeted at one exact model. What I saw with things like unified Invensense IMU drivers is that [...]

Maybe I was communicating in a bad way earlier. I do agree with you that there should be a driver specific for a single device and not make it handle the different devices. That is exactly what we are pushing for. 

Currently batt_smbus has been _overloaded_ with more stuff than what default SMBUS supports. I'm suggesting to split that into a bq40z80 (or with a bit more effort bq40zx0 specific driver) that the user can choose. I dont see why it should be a single unified driver that handles all SMBUS derived devices.

echoGee, 2021-01-26:
What do you think of this style ? 
```cpp
I2CSPIDriverBase *BATT_SMBUS::instantiate(const BusCLIArguments &cli, const BusInstanceIterator &iterator,
				      int runtime_instance)
{
	SMBus *interface = new SMBus(iterator.bus(), cli.i2c_address);
	if (interface == nullptr) {
		PX4_ERR("alloc failed");
		return nullptr;
	}

	int32_t batt_t;
	param_get(param_find("BAT_SMBUS_MODEL"), &batt_t);

	BATT_SMBUS *instance;

	switch(batt_t) {

		case 0:
		instance = new SMB_v11(iterator.configuredBusOption(), iterator.bus(), interface);
		break;

		case 1:
		instance = new BQ40Z50(iterator.configuredBusOption(), iterator.bus(), interface);
		break;

		case 2:
		instance = new BQ40Z80(iterator.configuredBusOption(), iterator.bus(), interface);
		break;

		case 3:
		instance = new Rotoye_Batmon(iterator.configuredBusOption(), iterator.bus(), interface);
		break;
               // Whats default ? 
		default:
		instance = nullptr;
		break;
	}
```

dagar, 2021-01-26:
> I dont see why it should be a single unified driver that handles all SMBUS derived devices.

Agreed!

> What do you think of this style ?

My preference would be each concrete BMS type be a separate module/driver. That way different boards can include only a subset of what they care about, start something specific, etc.

I might need to review some of the details, but shouldn't BATT_SMBUS simply be an extension of I2C that implements the smart battery data specification? Then each of these BMS (BQ40Z50, BQ40Z80, Rotoye_Batmon) would be a standalone driver module that HAS a BATT_SMBUS (physical) interface.

echoGee, 2021-01-26:
> My preference would be each concrete BMS type be a separate module/driver. That way different boards can include only a subset of what they care about, start something specific, etc.
> 
Could you explain/rephrase that? I'm not super thorough with pixhawk/architecture lingo.

> I might need to review some of the details, but shouldn't BATT_SMBUS simply be an extension of I2C that implements the smart battery data specification? Then each of these BMS (BQ40Z50, BQ40Z80, Rotoye_Batmon) would be a standalone driver module that HAS a BATT_SMBUS (physical) interface.

How would the function declaration look like if this is the case ?

BazookaJoe1900, 2021-01-26:
@echoGee the problem for me now, is to understand how different is the functionality of BatMon vs BQbq40zx0 vs solo
does on have more options then the other, for example reading more temperature sensors?
does this functionality is something you need, and implemented on BatMon. for example, I did changed the BQ driver on my local repo to read more temperature sensors.
How does thous changes implementations change the current driver,  addons? is that breaking changes? and how deep they break compatibility? are they changing the interface (uorb/mavlink)?

if the difference between BQ40Z50 and BQ40Z80 is only two if/else (this is the current state). I think they should stay unify.

if the difference to BatMon is much more, it can have another driver, that can be called BATT_SMBUS_ROYTOY_BATMON. it can be start from coping BAT_SMBUS. that approach will give you much more flexability to change BATT_SMBUS_ROYTOY_BATMON without any need to change interface of some "higher" BATT_SMBUS

dagar, 2021-01-26:
> > My preference would be each concrete BMS type be a separate module/driver. That way different boards can include only a subset of what they care about, start something specific, etc.
> 
> Could you explain/rephrase that? I'm not super thorough with pixhawk/architecture lingo.

Separate standalone modules for each. For example the Invensense IMU drivers. https://github.com/PX4/PX4-Autopilot/tree/master/src/drivers/imu/invensense

 - src/drivers/smbus/bq40zx0 (I don't yet have an opinion about a combined or split BQ40Z50 and BQ40Z80)
 - src/drivers/smbus/rotoeye
 - etc

Refactor truly common code (Smart Battery Data Specification).

echoGee, 2021-01-26:
Theres cleaning up to do, but what do you think of this structure? https://github.com/rotoye/px4_firmware_batmon/tree/batmon_4.03/src/drivers/batt_smbus 

- The files should be moved to their individual folders to be clearer. 
- Note that SBS 1.1 spec does not specify a lot of the register addresses that are used often such as the cell voltage. The core BATT_SMBUS now contains only the truly SBS spec registers.

nbelanger99, 2021-01-27:
> @echoGee the problem for me now, is to understand how difference of the functionality of BatMon vs BQbq40zx0 vs solo
> does on have more options then the other, for example reading more temperature sensors?
> does this functionality is something you need, and implemented on BatMon. for example, I did changed the BQ driver on my local repo to read more temperature sensors.
> How does thous changes implementations change the current driver, addons? is that breaking changes? and how deep they break compatibility? are they changing the interface (uorb/mavlink)?
> 
> if the difference between BQ40Z50 and BQ40Z80 is only two if/else (this is the current state). I think they should stay unify.
> 
> if the difference to BatMon is much more, it can have another driver, that can be called BATT_SMBUS_ROYTOY_BATMON. it can be start from coping BAT_SMBUS. that approach will give you much more flexability to change BATT_SMBUS_ROYTOY_BATMON without any need to change interface of some "higher" BATT_SMBUS

Hi @BazookaJoe1900, I'm working with @echoGee. The if-elseif structure works fine for differentiating between the BQ40z80/50 since there are only a few minor differences, and for now I've preserved that in [our new implementation](https://github.com/rotoye/px4_firmware_batmon/tree/batmon_4.03). However, the more platforms we want to support, the more hacky this kind of structure would get. However, we can abstract the default, generic SMBus use case which conforms to the SBS 1.1 spec and should therefore work for all SMBus-compatible devices, and build on top of that for extra features. For example, Batmon offers all cell voltages, and a seperate internal and external temperature reading (the latter feature isn't implemented in the above branch yet). 

We did have a [branch ](https://github.com/rotoye/px4_firmware_batmon/tree/batmon_v2.02_px4_v1.11)where we used this if-else structure before it was actually used to add BQ40z80 support, and even as such the source file is long and not very readable- it would only be worse as support is added for more smart battery types.  

As there's already a param for selecting specific battery models, I think it would make sense that we instantiate either the generic BATT_SMBUS driver or a more specific driver depending on what is used.

nbelanger99, 2021-01-27:
> Agreed!
>
>
> My preference would be each concrete BMS type be a separate module/driver. That way different boards can include only a subset of what they care about, start something specific, etc.
>
> I [...]

Hi @dagar, we've done something like this [here](https://github.com/rotoye/px4_firmware_batmon/tree/batmon_4.03). The idea is that BATT_SMBUS becomes a base class which can either be instantiated as a generic driver for any batteries which conform to the SBS spec, and more specific implementations are built on top of that. I haven't factored the BQ40z80/50 driver into separate classes yet, but that may be useful too?

dagar, 2021-01-27:
@nbelanger99 that sounds like roughly the right direction (in my view). I actually want to go a step further though to keep things modular with BQ40Z50, ROTOYE_BATMON, etc being separate modules. 

``` Console
nsh> bq40z50 start -I 2 # start bq40z50 on internal I2C2
nsh> rotoye_batman start -X 3 # start rotoye_batmon on external I2C3
```

People that use PX4 for custom builds can then trivially include only the specific BMS they need.

echoGee, 2021-01-27:
Would that mean the user would have to recompile a particular module and the default , lets say v1.12 wouldn't have all the modules. 

So the step that the user would have to do is add the module into a modified firmware, upload it and connect the hardware ?

echoGee, 2021-01-27:
Capturing some of the discussion we had with @dagar on a dev call. He suggested to move to a system where each device is a different module. He also suggested to keep the existing module so as to not break anything for existing users.  
Something similar to this. 
```
nsh> batt_smbus start ..... 
nsh> bq40z50 start -I 2 # start bq40z50 on internal I2C2
nsh> rotoye_batmon start -X 3 # start rotoye_batmon on external I2C3
```

This could be handled with a sub-optimal solution by potentially creating a core SBS1.1 only base class and using it in the rotoye_batmon module, while the batt_smbus exists as is(with the duplicated code). The batt_smbus would be improved later to remove duplicate code and split the bq40z50 into a different module later.

--------------------
However, after an internal discussion this seems that this intermediate sub-optimal solution could be pretty confusing to users. For eg, this is the parameters that are on QGC currently and the BAT_SMBUS_MODEL corresponds to different devices
[image]

However, if we implement the method suggested above, we would have this BAT_SMBUS_MODEL and an additional rotoye_batmon module creating confusion. 

From our perspective, we'd want BatMon to be plug and play after a simple parameter change. Therefore, it is my suggestion that we add another number(3 to select BatMon) to BAT_SMBUS_MODEL and handle it currently since it would be huge concerted effort to make an _ideal_ fix for this. I do understand that its not the best solution for the common good, but 

1. Having more than one parameter to handle the different devices, or conflicting information in different modules is too confusing imho. 
2. We do not have the resources to get ok from all the users using batt_smbus for a breaking change.

BazookaJoe1900, 2021-01-28:
I think @echoGee approach for one param that rule them all is correct.
also using hierarchy correct batt_smbus that has different implementation for each battery type sound good.
now, there are few point of decisions:
- which of the functionality is on the base class and which on de diverted. because there is a lot of shared code. one option is to duplicate code, and this option I like less.
- how to build correct hierarchy between the bqz50 and bqz80?

I looked at your branch, its need some more work to review it correctly and give constructive notes

echoGee, 2021-01-28:
> * which of the functionality is on the base class and which on de diverted. because there is a lot of shared code. one option is to duplicate code, and this option I like less.
From a cleanliness standpoint, since the name of the driver is smbus, we put the core SBS1.1 specs on the base class. There would be some duplication of code. But the duplication is for register names etc which could be differently handled on different devices. 
> * how to build correct hierarchy between the bqz50 and bqz80?
Currently it could be handled in a single class with a if function. If it diverges too much the structure is present to either 1.easily add a base bqzx0 class with common code and make z50 and z80 specific functions or 2. to make z50 and z80 specific functions
> 
> I looked at your branch, its need some more work to review it correctly and give constructive notes
I agree, but please give more details so that we are on the same page and we can work in the right direction.

nbelanger99, 2021-02-01:
I've opened a PR: #16723 related to this issue

FrankvVeelen, 2021-05-22:
Is there any more progress on this topic? I'm looking to implement the [isl94202](https://www.renesas.com/us/en/products/power-power-management/battery-management/multi-cell-battery-balancing/isl94202-standalone-3-8-cell-li-ion-battery-pack-monitor) chip's protocol, which I think is almost completely different from the ones mentioned here (and in fact uses I2C instead of SMBus). I think since a lot of different battery management chips have come out recently or are coming out soon-ish the more clean solution mentioned in this thread should be reconsidered. I have however not worked with the PX4 codebase, so any input is appreciated.

echoGee, 2021-05-22:
@FrankvVeelen There is a PR on this work: https://github.com/PX4/PX4-Autopilot/pull/16911
SMBus is an application layer protocol based on I2C

mhkabir, 2021-05-26:
Another PR, with improved structure to support even easier extension of SBS-based batteries: https://github.com/PX4/PX4-Autopilot/pull/17658