PX4/PX4-Autopilot #16113: V1.11.1: GPS over UAVCAN not detected

ThomasRigi, 2020-11-03:
**Describe the bug**
The GPS+Magnetometer unit (Here 2) does not get recognized over the CAN bus upon boot up on a Pixhawk 4 with v1.11.1 and UAVCAN enabled.

Unplugging and replugging the GPS unit without rebooting the Pixhawk makes it magically appear. 

**To Reproduce**
Steps to reproduce the behavior:
1. Switch the Here 2 unit to CAN mode and connect it to the Pixhawk 4 over either CAN1 or CAN2 port
2. Power the Pixhawk 4. Enable UAVCAN by setting UAVCAN_ENABLE to either automatic or manual config (1 or 2)
3. Reboot the Pixhawk
4. Observe the missing GPS. For example, in QGC's Mavlink Inspector there's no GPS_RAW_INT message and typing "uavcan status" in the mavlink console shows all empty in the gnss and mag fields.
5. Unplug and replug the GPS 
6. Wait a few seconds. You should now see the GPS_RAW_INT messages appear in QGC. "uavcan status" in the mavlink shell will also have found the GPS unit.

**Expected behavior**
The device should be discovered directly upon first boot without having to physically replug it.

**Log Files and Screenshots**
I don't know how to get a useful log of this. Tell me how to do it and I will upload it.

**Drone (please complete the following information):**
- Pixhawk 4 with PX4 v1.11.1
- Here 2 GPS unit set to CAN mode

**Additional context**
Was a working setup with v1.10.1

dagar, 2020-11-03:
Is there any other indication of what's going on? Do you have a Zubal Babel to see traffic on the CAN bus?

Does the unit have a fixed node ID or are you using dynamic node ID allocation?

amikhalev, 2020-11-03:
I observed something similar to this with my Here3 and Cube Orange a while ago.
I haven't seen it in a while though. Have you tried disabling serial GPS? (set `GPS_1_CONFIG` to disabled)

amikhalev, 2020-11-04:
I just observed this same issue on v1.11.1 with Here3 and Cube Orange. Replugging the CAN wire fixed it.

dagar, 2020-11-04:
> I just observed this same issue on v1.11.1 with Here3 and Cube Orange. Replugging the CAN wire fixed it.

Any difference relative to master?

amikhalev, 2020-11-04:
> Any difference relative to master?

On master it has been much more reliable for me. I had a crash today which included losing the Here3 but I think that is unrelated.

ThomasRigi, 2020-11-04:
I found a Here 3 in our stocks, same issue. By now I have tested with two different Here 2 units and one Here 3 on two different Pixhawk 4. All combinations were working on v1.10.1 and are not working on v1.11.1. We have a Babel, which sees the GPS and to my knowledge everything seems fine. There are gnss fix messages being sent. I really don't think it's a hardware issue.

I think the Here 3 has still dynamic node ID, but on one of the Here 2 we changed it to a static one. Nothing changed, independent of UAVCAN_ENABLE and GPS_1_CONFIG. 

Some additional tests I tried on the MAVLink Console in QGC: 
- gps stop, followed by gps start after some time: No reaction
- uavcan stop, followed by uavcan start after some time: no reaction. Tested both with and without running "gps stop" first.

The only thing to have the GPS messages appear in PX4 that I found was replugging the CAN cable. Which is very unpractical for us as the Pixhawk is not accessible when the drone is fully assembled to fly.

I just tested with master, seems to work. Would be good to be able to pin down which commit it was and get it into v1.11.2 :)

ThomasRigi, 2020-11-20:
@dagar I did some digging as on v1.11.2 it's still not working. I could track down until between commit 720784b9e9 (not working) and 458420f9cbaa55b84243d2f13f2e69d9236c37d8 (working). Build fails in commits in between with super long error messages that I have no idea what they are about. They are this massive bunch of commits about Updates for NuttX 9.1.0+ External schedule_note by you and @davids5. 

Is there a way to backport these changes?

ThomasRigi, 2020-11-30:
I read that a new bugfix release of 1.11 is on its way. It's important for us that this issue gets fixed for this new release as we want to move more sensors to the CAN bus, but it's worrying if the support breaks from one stable release to the next.

If it helps I can also bisect the other way around to see from where in v1.10.1 to v1.11.1 this broke. On v1.10.1 we were always using the GPS over UAVCAN without problems.

ThomasRigi, 2020-11-30:
The commit that broke it is https://github.com/PX4/PX4-Autopilot/commit/b399698306e5afa068903adda094adedf0423691

amikhalev, 2020-11-30:
Not sure about the commit that broke it, but I am guessing that what fixed it since v1.11 is changes the NuttX. [This commit](https://github.com/PX4/NuttX/commit/648ebc9ea9017440e55d41881b7ce8d32265be58) is the only thing I found after a quick search which could affect it.

ThomasRigi, 2020-11-30:
I tried to cherry-pick the changes of this file onto v1.11.2's version of NuttX, but it didn't help. :(  

I agree with you that it's most likely a change in NuttX that fixed it. Problem is I don't understand anything about it, RTOS are way beyond my knowledge :exploding_head:

davids5, 2020-12-01:
@ThomasRigi It is not surprising  that the CP did not work UAVCAN uses it's own driver not the NuttX one. There may be some clever-ness on the module, that is getting confused by the delay in the startup of UAVCAN, as an experiment you try to init the can controller sooner init time.

ThomasRigi, 2020-12-01:
@davids5 Can you please be more explicit on what exactly I could change where? Sorry, I'm a complete hardware and drivers noob :/

davids5, 2020-12-02:
@ThomasRigi it might be to great of a lift then. I do not have the HW so maybe @dagar can have a look?

ThomasRigi, 2020-12-07:
In https://github.com/ThomasRigi/Firmware/tree/uavcan-gps-v1.11.2 I moved the uavcan start before the board sensors initialisation in rcS and now it works :) Could there be any unwanted consequence of changing the startup sequence here?

And how would I create a PR for the new v1.11 point release announced in https://discuss.px4.io/t/px4-dev-call-november-25-2020/19652? A PR for master doesn't make a lot of sense I guess since the problem this commit fixes doesn't exist anymore

ThomasRigi, 2020-12-09:
Never mind, the fix actually only works with the SD card that comes shipped with the Pixhawk 4 and not with any others I tested (including SanDisk Extreme U3 32GB). Something with the initialisation must be really problematic if it depends on the SD card type.

Also, somehow I have killed the USB port of two Pixhawk 4 in the process and potentially other things inside them too. I think it was an SD card that managed to kill the USB port as I used the same SD card in both Pixhawks when they died, but I have no idea how this is even possible. Still, it's my best explanation of what happened.

And now I abandon this issue for good before it causes more damage to our parts.

LorenzMeier, 2020-12-09:
Given what you're seeing it is unlikely this is SD card or USB related. I would review your whole setup - in situations like this one tends to swap parts around without realizing the one common thing we're stilling using. E.g. a power supply, a bad FTDI cable, a logic analyzer or briefly mounting the units into an airframe that has a short to battery voltage, etc.

closed 2020-12-09 by LorenzMeier