PX4/PX4-Autopilot #16114: Incorrect Fused Altitude Estimate with blocked LIDAR-Lite v3

eli-IF, 2020-11-03:
**Describe the bug**
While flying with the Lidar-Lite V3 enabled, if there is anything blocking the sensor the EKF2 Altitude estimation drifts from the correct value. This means the vehicle will not hold altitude in Position or Altitude modes.

**To Reproduce**
Steps to reproduce the behavior:
1. The parameters that affect behavior are SENS_EN_LL40LS, 2 and EKF2_RNG_AID, 1
2. Take off in position control mode with the lidar lite v3 covered so the reading does not change
3. After about a minute the fused altitude estimate will diverge from the GPS and barometer
4. The drone becomes very difficult to control in Position mode so switching to Manual mode is necessary

**Expected behavior**
I expected the drone to fly normally and listen to the commanded stick inputs. I expected that the lidar-lite values would be ignored if they did not fall within a reasonable threshold so that they did not ruin the EKF2 altitude estimate.

**Log Files and Screenshots**
Flight review: https://review.px4.io/plot_app?log=5053dee9-6ea4-45bc-9ef8-cef084c4c192
[image]
Here you can see that the fused altitude estimate is diverging from both GPS and barometer. 

**Drone (please complete the following information):**
The drone is a custom built hexacopter and has a Modal-AI flight core. The Lidar-lite v3 is wired through i2c and is a Garmin https://buy.garmin.com/en-AU/AU/p/557294

**Additional context**
This issue also occurs when something swings under the lidar-lite while the drone is flying. The lidar-lite does not need to be covered the entire time to get a bad altitude estimate. The drone will only descend at the landing speed when this situation occurs.

bresch, 2020-11-06:
[image]
The reason it's still fused is that the reported quality is good, the value isn't stuck and it is the only source of height fused at that time. To detect this issue we would need to monitor the GPS and baro heights when fusing range height or to fuse all height sources and we should be able to detect the error it in the innovation.

eli-IF, 2020-11-06:
Okay, so it seems like there is no immediate solution except turning off the lidar-lite when a payload is large enough to cover it? Can you point me in the right direction to make the changes you discussed? I would be happy to make some changes and submit a PR but I don't really know where to start

dagar, 2020-11-07:
> Okay, so it seems like there is no immediate solution except turning off the lidar-lite when a payload is large enough to cover it? Can you point me in the right direction to make the changes you discussed? I would be happy to make some changes and submit a PR but I don't really know where to start

How about a custom minimum for valid data?

eli-IF, 2020-11-07:
> How about a custom minimum for valid data?

That sounds like a good quick fix. Where would that go? here? https://github.com/PX4/PX4-Autopilot/tree/master/src/drivers/distance_sensor/ll40ls

eli-IF, 2020-11-07:
Actually maybe adding a little more logic to this if statement would fix this issue? https://github.com/PX4/PX4-Autopilot/blob/master/src/lib/drivers/rangefinder/PX4Rangefinder.cpp#L88

dagar, 2020-11-07:
> Actually maybe adding a little more logic to this if statement would fix this issue? https://github.com/PX4/PX4-Autopilot/blob/master/src/lib/drivers/rangefinder/PX4Rangefinder.cpp#L88

Yes, that works as a hack if you want to test it. If that works for you then something like an EKF2_RNG parameter would probably be fine.

eli-IF, 2020-11-10:
After looking at it a bit further, would increasing the report.min_distance through the EKF2_MIN_RNG do exactly the same thing as the hack you suggested?

dagar, 2020-11-10:
> After looking at it a bit further, would increasing the report.min_distance through the EKF2_MIN_RNG do exactly the same thing as the hack you suggested?

Not quite, that's an offset for what the distance sensor would be reading when you're actually on the ground (eg landing gear). I suppose you could make an argument that any reading within that distance is bogus.

What you want is the ability to effectively adjust the min distance at runtime.

eli-IF, 2020-11-10:
My confusion stems from how I would tell the estimator that this is a bogus value. I was going to just set the signal quality to zero which is done in that if statement already. 

Is there a different way to tell EKF2 to ignore the value?

BazookaJoe1900, 2021-04-26:
@dagar @eli-IF @priseborough 
The problem might be more common than you think (I got similar issue on the past). 
on my case the Lidar (same Lidar lite v3) sent what seems as valid readings of close range though the drone was on air.

Is it possible to add mechanism that disable the use of the Lidar if it been rejected by the EKF?
At least for the rest of the current flight.
That might should be enabled by parameter, and might involve thresholds/timeouts arguments

ThomasRigi, 2023-01-16:
It's still an issue on v1.13 : https://logs.px4.io/plot_app?log=9e85c01f-4d35-4639-a6ae-b3dcb0d511fa
[image]

The drone climbed over 200m thinking it was still basically on the ground... This should not happen ! Especially with the well-working GPS set as the primary source of height (EKF2_HGT_MODE = 1). 

Would be really good to have this fixed.