PX4/PX4-Autopilot #16122: Fail to trigger the position (GPS) loss failsafe

KimHyungSub, 2020-11-05:
**Describe the bug**

While I have experimented the GPS loss failsafe, I have found that PX4 fails to trigger the failsafe mode.
I think this issue is related to #15065. Thanks.

**To Reproduce**
Steps to reproduce the behavior:
Frist case, 
- make px4_sitl jmavsim
- Takeoff and fly (e.g. using Joystick in QGC)
- Enable failure injection: param set SYS_FAILURE_EN 1
- param set COM_POS_FS_DELAY -1
- Turn on orbit mode
- failure gps off
- fail to trigger the GPS failsafe
* Result: keep floating in the air

Second case,
- make px4_sitl jmavsim
- Takeoff and fly (e.g. using Joystick in QGC)
- Enable failure injection: param set SYS_FAILURE_EN 1
- param set COM_POS_FS_DELAY -1
- flying into another location (i.e., by using "go to the location" in QGC)
- failure gps off
* Result: blind land

If the flight mode is another mode except orbit or "go to the location", PX4 well triggers the GPS failsafe even though the parameter (COM_POS_FS_DELAY) has a strange value. To fix such issue, I think that PX4 needs to check the valid range of the parameter or something else.

**Expected behavior**
The GPS failsafe is triggered.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=a5823c45-07e5-43f5-9227-e8493ae89e54
https://logs.px4.io/plot_app?log=05fd86e9-3f3f-48c7-aa4f-117894ce26a3

KimHyungSub, 2020-11-07:
Update: this problem is come from getting rid of valid range check of the parameter. To be specific, PX4 v.1.7.2 forces the parameter to have a value in the valid range (https://github.com/PX4/PX4-Autopilot/commit/dc2d6e8aab4148f4672e3cf674c7e85f86cf5533#diff-ddb86f30e5ea1d95166ac519fb5ab96535e9e43a4df3b2b1e32668d1ca4d1195). 
However, the current PX4 version just uses the parameter without checking the value of the parameter.

LorenzMeier, 2020-11-08:
The parameter system already prevents you from setting invalid parameters and the range is clearly specified correctly (1-100 seconds). If you force an invalid parameter, horrible things can happen. This is true as well for pretty much any other area of the configuration system: If you force a value outside of the valid range, things can fail.

Configuration management is as important as correct code and if you set parameters without checking the range or you force an override in QGC, then you do the same as changing code incorrectly.

closed 2020-11-08 by LorenzMeier