PX4/PX4-Autopilot #16123: Wrong displayed altitude

KimHyungSub, 2020-11-05:
**Describe the bug**
Could you guys check and solve the issue?
https://discuss.px4.io/t/wrong-altitude-display-on-qgc/17261

**To Reproduce**
Steps to reproduce the behavior:
- make px4_sitl jmavsim
- Takeoff without any RC Joystick
- Enable failure injection: param set SYS_FAILURE_EN 1
- failure gps off
- PX4 conducts the blind landing, but QGC keeps showing the last reported altitude from the GPS.

**Expected behavior**
PX4 needs to send correct GLOBAL_POSITION_INT MAVLink messages according to decreased altitude during the blind landing.

Jaeyoung-Lim, 2020-12-07:
@KimHyungSub Could you clarify why the expected behavior to keep sending GPS messages after the GPS has failed?

Wouldn't the more "correct" behavior be that the vehicle stops sending the GPS messages and QGC displays that accordingly?

KimHyungSub, 2020-12-08:
@Jaeyoung-Lim Thank you for the feedback! This is because QGC expects that PX4 keeps sending GLOBAL_POSITION_INT messages to GCSs under GPS loss, as explained in the link (https://discuss.px4.io/t/wrong-altitude-display-on-qgc/17261). However, in my opinion, this problem only happens on SITL. I tried to trigger this issue on a real drone, but, I couldn't.

Jaeyoung-Lim, 2020-12-09:
> This is because QGC expects that PX4 keeps sending GLOBAL_POSITION_INT messages to GCSs under GPS loss, as explained in the link (https://discuss.px4.io/t/wrong-altitude-display-on-qgc/17261). 

Right, but that doesn't make the correct behavior of the firmware to keep sending gps messages even though it is not getting a GPS fix. 

> However, in my opinion, this problem only happens on SITL. I tried to trigger this issue on a real drone, but, I couldn't.

Can you confirm that the SITL vehicle and your realdrone has the exact same configuration? If not, this might be coming from configuration differences.

LorenzMeier, 2020-12-16:
The autopilot is continuing to send the ALTITUDE message and it would in fact be *incorrect* to keep sending a position that is not any more valid.

closed 2020-12-16 by LorenzMeier