PX4/PX4-Autopilot #16129: Cube Black does not recognize Here2 or Here3 over CAN

ValentineStone, 2020-11-05:
**Describe the bug**
A Cube Black running PX4 does not recognize either Here2 or Here3 connected to CAN port.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash Cube Black's firmware to either PX4 1.11.1 or 1.10.1
2. Connect Here2 or Here3 to CAN1 or CAN2 port
3. See Here2/3 lights turn blue before quickly turning off
4. Go to MAVLink Inspector
5. See `GPS_RAW_INT` messages with `fix_type` 0 - `GPS_FIX_TYPE_NO_GPS`

**Expected behavior**
- `GPS_RAW_INT` messages should have `fix_type` other than 0, since there is GPS module connected
- Here2/3 lights should be flashing green

**Log Files and Screenshots**

`GPS_RAW_INT` with status of no GPS found

[image]

MAVLink console, `uavcan status`. Sorry, could not figure out how to get this as file or text

[image]
[image]
[image]

**Drone:**
- Just the cube and gps modules

**Additional context**
- This is probably not #16113, for I tested with both px4 1.10.1 and 1.11.1, and neither worked.
- This also is likely not linked to Here2-switching-to-CAN-mode-debacle like in #13609, my Here2 is recent, has the latest firmware and CAN connector out of the box (its rev 2.2)
- I also tested with both static and dynamic Node ID for both GPS modules, as outlined here https://docs.cubepilot.org/user-guides/here-3/here-3-manual#2-using-px4-firmware , did not seem to help.
- Might be something like what @mwiatt was having in #12592 - Here2 not being detected when using CAN, don't know if they got it working since then though.

mattpaddock, 2020-11-06:
I've used the Here2 on 1.10.1 over CAN with no issues. Have not tried with 1.11.1, but Here2 does not work properly with 1.11.1 on I2C either (https://github.com/PX4/PX4-Autopilot/issues/16035). 

@ValentineStone Just to mention the most common reason I've seen people have issues with Here2 CAN... the internal switch on the Here2 board is set to CAN mode and not I2C? Some come from factory in CAN mode and some come from factory in I2C mode. I've got about ~40 of the Here2's and I swear they change every time.

[image]

JacobCrabill, 2020-11-06:
@ValentineStone the output of `uavcan status` that you posted is showing that the Here2 is working properly.  You'll need to look at the uORB data being published to see more details (`listener sensor_mag`  /  `listener vehicle_gps_position`)

ValentineStone, 2020-11-17:
@mattpaddock Yeah, I disassembled the module and checked for that, I have a rev 2.2, it does not have that issue.
@JacobCrabill Well, turns out you are right, it is working.

I was assuming that `fix_type` in [`GPS_RAW_INT`](https://mavlink.io/en/messages/common.html#GPS_RAW_INT) messages confirmed to the [`GPS_FIX_TYPE`](https://mavlink.io/en/messages/common.html#GPS_FIX_TYPE) enum, and 0 means no GPS connected:

[image]

A colleague of mine who has way more experience with px4 than me just brought the thing outside, where it would be able to acquire GPS fix, and it did, despite the 0 in `fix_type`.

So, GPS is working, but px4 does seem to not confirm to the MAVLink spec here, I might create an issue for that instead, not sure if I it is something deserving of an issue though.

closed 2020-11-17 by ValentineStone

mrpollo, 2021-03-02:
@ValentineStone did you ever create the issue mentioned above?

ValentineStone, 2021-03-03:
@mrpollo No, never got around to it, my company switched to Ardupilot for external reasons, so I didn't have any time to spend on PX4 since then.