PX4/PX4-Autopilot #16132: FW Position Control Landing Glideslope Faceing South Causes Vehicle Crash

AlexanderAurora, 2020-11-05:
**Describe the bug**
If you set up a mission with a fixed-wing landing pattern facing south. There is a chance the vehicle will crash itself by nose-diving into the ground. I was using Firmware v1.10.2 but it looks like this is still and issue on v1.11.1 and your master branch.

**To Reproduce**
Steps to reproduce the behavior:
1. Plan simple fixed-wing mission with landing facing almost directly south.
2. Upload the mission to real vehicle or simulator.
3. Make sure you have a gusty crosswind that blows the vehicle off course during the landing.
4. Fly the mission and allow for autonomous landing.

Screen of the simulated mission I used to reproduce the issue.
[image]

**Expected behavior**
During the glide-slope portion of the landing sequence, the fixed-wing controller will immediately set the altitude command to [terrain_alt + 0] and crash.

**Log Files and Screenshots**
Plot of altitude setpoint and altitude estimate during a south-facing landing when the bug occurred: 
[image]

Plot of same data from a non-south-facing landing.
[image]

This was caused by a conditional in the getLandingSlopeRelativeAltitudeSave(float, float, float) function.
This function compares the bearing from the previous waypoint to the landing point, and the bearing from the aircraft to the landing point. If the difference between these two bearings is >=90 it returns zero. 

src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp (where the landing slope function is called)
[image]

src/lib/landing_slope/Landingslope.cpp:
[image]

I think the idea behind returning zero is that if difference in those bearings is >90, we overshot our landing target and we should get on the ground ASAP.

src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp (where the two heading arguments are calculated)
[image]
[image]

the function get_bearing_to_next_waypoint() is in src/lib/ecl/geo/geo.cpp. It wraps the result from -pi to pi.

In the case where the landing is positioned almost directly south, bearing_lastwp_currwp = +3.13
If the vehicle gets blown slightly off course to the east, bearing_airplane_currwp = -3.13
abs(3.13 - (-3.13)) = +6.26 > 1.57 even though 3.13 and -3.13 are almost the same heading.
This results in: altitude_desired = terrain_alt + 0

The figure below shows the values of bearing_lastwp_currwp (bpc) and bearing_airplane_currwp (bac) and the value of the sloped altitude command. This was recorded during a simulation.
[image]

A simple fix for this would be to apply wrap_pi to the result of the heading subtraction in Landingslope.cpp.
[image]

[image]

I think you should steer clear of subtracting angles from one another, id say use the arccos(dot(A,B)) to find the angle between two unit vectors. this always gives you a value from 0-180 deg.

bresch, 2020-11-06:
Thanks for the investigation.

> A simple fix for this would be to apply wrap_pi to the result of the heading subtraction in Landingslope.cpp.

Could you make a PR with that wrap_pi and link it to this issue please?

All additions/subtractions of angles should always be in a `wrap_pi` to be safe. Taking arccos of the dot product of two unit vectors gives the same result but is way more computationally expensive (2 sqrt, 2 divisions, 1 arccos, ...); except if we anyway need the unit vectors for something else.

ryanjAA, 2020-11-10:
If this has the ability to bring down a plane that has a seemingly well planned mission, wouldn’t fixing this (and anything else incremental out there) be prudent and version bumping 1.11.1 to 1.11.2? Seems like this is something that could have serious unintended consequences for users that are following the guides and doing things correctly. @dagar

AlexanderAurora, 2020-11-10:
Excuse my novice-ness here but it looks like in order to create a pull request, the branch with the changes needs to be pushed to the repo. I don't have the permissions level to push right now.

I can open a separate issue for this but it's such a small fix I'm unsure if it warrants it. There was another instance where using an autonomous hand-launch of a FW aircraft caused a nose-dive crash.

In the function TECS::_initialize_states() there is an "IF - ELSE IF" statement where either initialization can happen, or climbout limits set, but both cannot happen at the same time. 

[image]

As soon as the vehicle is launched, the tecs controller initializes and detects under-speed, but the climbout pitch limit is not set, and the aircraft sets a high negative pitch value for one iteration. This was enough to cause the vehicle to nose-dive. 

This was fixed on my end by making the "IF - ELSE IF" into two separate IF-statements so that climbout limits can still be set on initialization.

closed 2020-11-10 by dagar

ryanjAA, 2020-11-11:
@AlexanderAurora I can only get this to crash if I plan a landing at exact 180º S. 181 won't let it crash and neither will 179,  359, 360 and zero (same) also don't let it crash in SITL. 

Were you seeing something different?

PR #16158 (2020-11-10): Afs/landing slope dive
Please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/) to align on pull requests if necessary. You can then open draft pull requests to get early feedback.

**Describe problem solved by this pull request**
 Functions in the Landingslope class evaluate heading differences without wrapping pi.

**Describe your solution**
Add wrap_pi to LandingSlope.cpp functions to protect against returning zero when we don't want to.

**Describe possible alternatives**
Use get_vector_to_next_waypoint() instead of get_bearing_to_next_waypoint() to get a two 2-d v [... cut]