PX4/PX4-Autopilot #16133: Hardfault on shutdown when CONFIG_DEV_CONSOLE not present

amikhalev, 2020-11-05:
**Describe the bug**
A hardfault occurs on shutdown.

**To Reproduce**
1. Upload PX4 master (I tested on bde5cab) to board without CONFIG_DEV_CONSOLE such as Cube Orange.
2. Run `reboot`
3. Observe hardfault [on this line](https://github.com/PX4/PX4-Autopilot/blob/bde5cadba7cbf1739e1c409773f9d8ff4bd24a59/platforms/common/shutdown.cpp#L176)

**Expected behavior**
No hardfault

**Log Files and Screenshots**
[image]

**Drone (please complete the following information):**
CubePilot CubeOrange + PX4 master bde5cab

**Additional context**
I think this is related to apache/incubator-nuttx#2203, but [the fix here](https://github.com/PX4/PX4-Autopilot/blob/master/platforms/nuttx/src/px4/common/px4_init.cpp#L45) does not apply since the shutdown function is being run in the `hpwork` task which is created based on the idle task before `px4_platform_console_init` runs.

danduril, 2021-03-24:
Perhaps a similar hard fault, here: https://review.px4.io/plot_app?log=e42e93ed-4877-4605-bbc7-f4f96544c5b4

closed 2021-03-24 by LorenzMeier

amikhalev, 2021-04-07:
I no longer see hard faults at shutdown. Thanks!