PX4/PX4-Autopilot #16135: The listener can't display the topic "gps_inject_data".

Ladsers, 2020-11-06:
**Describe the bug**
The listener command can't display the topic "gps_inject_data". All other topics are displayed correctly. Message error: "Topic did not match any known topics", but the topic "gps_inject_data" is in the general list of topics.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect pixhawk 4 via radio and here+ via usb.
2. Start QGroundControl.
3. Make sure that the "RTK" icon is present.
4. Open Mavlink console.
5. Type in console: `listener gps_inject_data`.
6. See error message: "Topic did not match any known topics".

**Expected behavior**
Get message containing the following fields (from "gps_inject_data.msg"):
uint64 timestamp		# time since system start (microseconds)
uint8 len			# length of data
uint8 flags         		# LSB: 1=fragmented
uint8[182] data		# data to write to GPS device (RTCM message)

**Screenshots**
[image]
[image]
[image]
[image]

dagar, 2020-11-06:
A small set of messages are excluded in listener because they didn't seem useful and the generated print code is rather large.

https://github.com/PX4/PX4-Autopilot/blob/91d1825fcf25bfe7f4f98379f0501bcde41ef51c/src/systemcmds/topic_listener/generate_listener.py#L16

I'm not opposed to removing `gps_inject_data` if you think it might be useful.

dagar, 2020-11-08:
I'll remove the excludes https://github.com/PX4/PX4-Autopilot/pull/16139.

closed 2020-11-10 by dagar

Ladsers, 2020-11-16:
I'm sorry for the long silence.
@dagar thanks! Now it works correctly.