PX4/PX4-Autopilot #16136: fatal: Not a git repository: --absolute-git-dir

Owen1u, 2020-11-06:
After gitting clone the last version of PX4 fireware and gitting submodule update,when I make px4_fmu-v2_default, it shows the following text:

[ 10%] Generating nuttx/include/nuttx/config.h, nuttx/include/nuttx/version.h
make[7]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[7]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
fatal: Not a git repository: --absolute-git-dir
make[7]: *** [/home/luminjun/PX4/Firmware/build/px4_fmu-v2_default/external/Build/px4io_firmware/NuttX/nuttx/.version] Error 1
NuttX/CMakeFiles/nuttx_context.dir/build.make:63: recipe for target 'NuttX/nuttx/include/nuttx/config.h' failed
make[6]: *** [NuttX/nuttx/include/nuttx/config.h] Error 2
make[6]: *** Deleting file 'NuttX/nuttx/include/nuttx/config.h'
CMakeFiles/Makefile2:6651: recipe for target 'NuttX/CMakeFiles/nuttx_context.dir/all' failed
make[5]: *** [NuttX/CMakeFiles/nuttx_context.dir/all] Error 2
Makefile:105: recipe for target 'all' failed
make[4]: *** [all] Error 2
src/drivers/px4io/CMakeFiles/px4io_firmware.dir/build.make:110: recipe for target 'external/Stamp/px4io_firmware/px4io_firmware-build' failed
make[3]: *** [external/Stamp/px4io_firmware/px4io_firmware-build] Error 2
CMakeFiles/Makefile2:6715: recipe for target 'src/drivers/px4io/CMakeFiles/px4io_firmware.dir/all' failed
make[2]: *** [src/drivers/px4io/CMakeFiles/px4io_firmware.dir/all] Error 2

so how can i solve the problem?

dagar, 2020-11-06:
What's your exact build environment (ubuntu version, etc)? You likely have a version of git that's slightly too old.

Owen1u, 2020-11-07:
> What's your exact build environment (ubuntu version, etc)? You likely have a version of git that's slightly too old.

ubuntu 16.04 
git version:2.7.4

dagar, 2020-11-08:
> > What's your exact build environment (ubuntu version, etc)? You likely have a version of git that's slightly too old.
> 
> ubuntu 16.04
> git version:2.7.4

If you update `git` it might work, but personally I would update to the current Ubuntu LTS as this point or start building within a docker container. We generally can only support the current Ubuntu LTS (20.04) and the previous (18.04).

closed 2020-11-08 by dagar