PX4/PX4-Autopilot #16143: BNO055 module?

IGNNE, 2020-11-09:
I would like to use the BNO055 IMU with PX4. Now, this IMU is special, since it already does the sensor fusion on chip and provides finished orientation data. I'm running on a RPi, since I need the power for some camera processing. I didn't find any info on using that IMU with PX4, though.

I used a ROS driver node in the past, maybe that could be connected to PX4? Or would it be better/easier to just convert the existing IMU driver into a module?

dagar, 2020-11-09:
I would implement a new driver for it, probably not that different than one of the existing Bosch drivers. https://github.com/PX4/PX4-Autopilot/tree/master/src/drivers/imu/bosch

You could have the device publish an attitude directly, but to be honest it's probably easier to simply publish accel/gyro and use the existing estimator.

IGNNE, 2020-11-09:
Thanks for the answer, I must admit I didn't consider using the IMU as separate sensors before. It's an interesting idea, since it allows for a pretty simple driver. 

On the other hand, I have to reinvent the wheel some way or another, since it doesn't make any sense to have two sensor fusion algorithms running. If I wanted to use the on-board MCU, would it be a good idea to write a PX4 "estimator" that essentially does nothing but passes the MCU output on?

dagar, 2020-11-10:
> If I wanted to use the on-board MCU, would it be a good idea to write a PX4 "estimator" that essentially does nothing but passes the MCU output on?

Basically you just need to publish the equivalent (eg `vehicle_attitude`) data and prevent the existing estimator(s) from doing the same. This would work fine for the simple cases where you only have an attitude estimate (SYS_MC_EST_GROUP 3), but quickly becomes more effort than it's worth when needing a position estimate. The default estimator ekf2 provides attitude, velocity, position, etc. It also estimates accel and gyro biases, which makes a huge difference for these cheap sensors. 

I would recommend you implement the BNO55 as a simple IMU (accel + gyro) driver.

dagar, 2020-11-10:
I believe this has been answered, feel free to contact me on slack (@dagar) if you have any specific questions.

closed 2020-11-10 by dagar