PX4/PX4-Autopilot #16165: Possible Bug:  pixhawk hanged and rebooted with v1.11.1

RamDG, 2020-11-11:
**Describe the bug**
A clear and concise description of the bug.

Hi,
What am I doing:   I am using  jetson nano  board with pixhawk cube black (fmuv_3) for object tracking. I am using the latest firmware v1.11.1 stable for my Hexacopter. i have created some parameters for object selection in a new module i have created. When i touch on video that particular point in the video will be selected as target point and the same is tracked by the algo running on jetson .  When i click on the object only once,  that  xyz location is written on the parameters i have created in the QGC parameter list.

Possible  Bug:  when i touch on the  video multiple times. CPU load goes to 100% and i start getting errors like accel_0 accel_1 failed  or mag failed.  Yesterday i flew  my Hexacopter. It was in the position hold mode. When i touched the vido to select the object for tracking mag failed and then the entire gps unit failed and the vehicle log says it tried for fail-safe but it crashed.  Here is a link for the log file:
https://logs.px4.io/plot_app?log=b78baa94-c48a-4f98-b032-5cd031f53cf6

After the crash i checked the same issue on v1.10.1 and ardupilot.  and the issue was not there.

Along with that this issue is alwyas there on v1.11.1
https://github.com/PX4/PX4-Autopilot/issues/16035

dagar, 2020-11-11:
Can you comment on what the custom `gimbal_control` module is doing on these changes? 

[image]

RamDG, 2020-11-11:
I have created a simple  1 axis gimbal  which has a laser pointer.... gimbal is free to move  in pitch direction only.
I am using aux  pins to connect 2 servos  on aux1 and aux2 to control the gimbal movement. One more servo on aux3 is being used to turn on and off the laser when i am tracking the object.  Also, all the tracking related parameters  are being accessed through this module.

FYI :  just for testing purpose i stopped this module and tried to write on  some other available parameter continuously through a loop.... then also i observed the same . CPU load goes to 100% and sensors starts failing one by one.

dagar, 2020-11-12:
> write on some other available parameter continuously through a loop

When you write a parameter all modules in the system are notified of the update and then they refresh their local copy. At a bare minimum you can use `param_set_no_notification()`, but really you should be using uORB sharing data between modules.

RamDG, 2020-11-13:
Suggestion:  I believe px4 must  not get hanged  in any case. At least when system is ARM. Therefore, a software level protection is required  !!!!!!!!!

dagar, 2020-11-13:
> Suggestion: I believe px4 must not get hanged in any case. At least when system is ARM. Therefore, a software level protection is required !!!!!!!!!

High rate parameter updates are a bit of an abuse of the system, but we can add protections. https://github.com/PX4/PX4-Autopilot/pull/16189

RamDG, 2020-11-26:
@dagar  I tried  param_set_no_notification() for the parameter that i am using for tracking which is GOD_S1.
Still every-time when i select  an object to track , cpu load goes to 100%. please let me know how it can be resolved.

https://logs.px4.io/plot_app?log=ca8d6da5-af38-4057-ba6b-2fdf78966b2a

Also in the watchdog i am seeing FD for gimbal_control is 10 and for other apps it is quite low. Can you please tell what is it representing.  and what is a nominal value for the same.
[image]