PX4/PX4-Autopilot #16179: Auto RTL/Land command is ignored when in offboard mode

smnogar, 2020-11-12:
**Describe the bug**
When the vehicle is being flown with offboard local position setpoints, if auto RTL or land is triggered, the vehicle switches into the expected mode but does not change its behavior until reaching the last sent setpoint. At that point it will begin to execute auto land or RTL. 

I know this is an unlikely series of events, but I am attempting to implement an RC failsafe while in offboard mode, and it seems that navigator is not correctly setting mission setpoints if coming from offboard mode. 

**To Reproduce**
Steps to reproduce the behavior:
1. Fly mission with local position offboard (sending x,y,z) offboard setpoints
2. In mavlink console, send command: "commander mode auto:land"
3. Vehicle continues to fly to last sent offboard setpoint

**Expected behavior**
Offboard should be dropped, and RTL or Auto land mode should be entered and the failsafe behavior begins.

**Log Files and Screenshots**
Log is at https://review.px4.io/plot_app?log=626f96e9-3301-4362-98e8-127e4301e693

**Drone (please complete the following information):**
- This was verified both in flight using modified firmware and in SITL simulation. 
- The vehicle continues to fly to the last sent setpoint even if no more offboard setpoints are being sent.

ThomasRigi, 2020-11-24:
We had the same problem with global position offboard setpoints: https://logs.px4.io/plot_app?log=cf12715b-a5bd-421e-9e9e-b46449691f8a. This was our fourth flight ever using global position offboard setpoints, so to me it's not an unlikely series of events. RTL is one of the failsafe options for offboard loss. And also if you manually trigger RTL or Land you probably have a good reason to do so, making it really important that the drone actually does what it's supposed to.

Also, it's not an offboard-only problem. Here's a flight log where the drone should have landed after a Quadchute in mission mode (one-line change with respect to v1.10.1) : https://logs.px4.io/plot_app?log=f66382bd-bcbd-4a78-88fe-577c178e1727. 
I really don't understand where the drone was heading to, but it was tracking the velocity setpoints reasonably well (55:02-55:12)...

And here on stock v1.10.1, RTL after Quadchute: https://logs.px4.io/plot_app?log=c0758441-61c8-40ff-9945-1206190d65c9
Again, no idea at all what the drone was doing from 16:18-16:26. Actually, until 16:42 to be precise as only at that point the drone directed itself towards the home / land spot.

I'm not 100% sure if the root cause is the same for the offboard behaviour and for the mission-quadchute behaviour, but to me it seems very likely. Both cases have bad position setpoints in Land & RTL.

BazookaJoe1900, 2021-01-08:
I think that the next PR solves the issue, but its after v1.11.3:
MulticopterPositionControl: Reset task between auto mode switches #16304

smnogar, 2021-01-08:
Yep, @benjinne verified that #16304 solves this. Closing.

closed 2021-01-08 by smnogar