PX4/PX4-Autopilot #16187: UART flow control (CTS/RTS)

TomRvr, 2020-11-13:
**Goal**
I want to enable hardware flow control on a TELEM port in order to connect it to a TTL to RS485 converter  

**Bug**  
RTS pin state doesn't change when flow control is activated  

**To Reproduce**  
When doing `mavlink status` I get :  
[image]

In `px4/fmu-v5/nuttx-config/nsh/defconfig` if I add the following to activate flow control on TELEM1 (/dev/ttyS1)   :  
```
CONFIG_USART1_IFLOWCONTROL=y
CONFIG_USART1_OFLOWCONTROL=y
```  
Then I can't connect to QGC anymore
  
I activate flow control on TELEM2 (/dev/ttyS2) doing `mavlink start -d /dev/ttyS2 -b 57600 -z -x` and then I get this  
[image]
So it looks like the activation worked

If I plug CTS to GND then I can see data on TX, otherwise there is no data on TX so this part seems to work
But, the state of the RTS pin never changes

**Expected behavior**
RTS pin should be high when transmitting, low otherwise

**Hardware**
[CUAV V5](http://doc.cuav.net/flight-controller/v5-autopilot/en/v5.html)  
[TTL to RS485 converter](https://www.amazon.co.uk/iHaospace-MAX485-Module-RS-485-Development/dp/B07L3KWF6W/ref=psdc_949408031_t4_B0785B3BS8)  
Here is the wiring that I want to do :  
[image]

I'm not sure yet if it's a bug or a configuration mistake on my side  
Any help would be appreciated, I can provide more details if needed  
Thanks

TomRvr, 2020-11-17:
I noticed the parameter **STM32_FLOWCONTROL_BROKEN**
In the [Kconfig ](https://github.com/PX4/NuttX/blob/904a602c74dc08a100b5c2bd490807de19e73e10/arch/arm/src/stm32/Kconfig) file, it says :    
```
config STM32_FLOWCONTROL_BROKEN
	bool "Use Software UART RTS flow control"
	default n
	---help---
		Enable UART RTS flow control using Software. Because STM
		Current STM32 have broken HW based RTS behavior (they assert
		nRTS after every byte received)  Enable this setting workaround
		this issue by using software based management of RTS
```  
And in my [defconfig ](https://github.com/PX4/PX4-Autopilot/blob/master/boards/px4/fmu-v5/nuttx-config/nsh/defconfig) I have **STM32_FLOWCONTROL_BROKEN=y**  

I also found **STM32_HCIUART_SW_RXFLOW** which looks quite similar  
```
config STM32_HCIUART_SW_RXFLOW
	bool "Use Software UART RTS flow control"
	default n
	---help---
		Enable UART RTS flow control using Software.  Current STM32 have
		broken HW based RTS behavior (they assert nRTS after every byte
		received)  Enable this setting workaround this issue by using
		software based management of RTS

		If HCI UART DMA is enabled, this is probably the better selection
		as well. In that case, the Rx DMA buffer will avoid Rx overrun due
		to short, bursty activity.  Software RTS management will probably
		result in overall better throughput and should still avoid Rx data
		overrun conditions.
```  
So it seems that there is a known bug on hardware flow control on STM32F7 with a workaround using software flow control  
But I'm not sure how to activate it properly  

Some parameters doesn't seem to be taken into account, when I set  
```
CONFIG_USART2_IFLOWCONTROL=y
CONFIG_USART2_OFLOWCONTROL=y
```
Then I see `flowcontrol: OFF` in `mavlink status` response

It looks like [this script](https://github.com/PX4/PX4-Autopilot/blob/master/Tools/serial/generate_config.py) overrides some of the parameters, at least the baudrate, could it be a problem ?

LorenzMeier, 2021-01-01:
We could provide further pointers if you would like to dive into the code - but I currently don't have hardware to reproduce / test. @davids5 might be able to provide pointers in NuttX.

Igor-AirQ, 2021-09-06:
We ran into the same problem when we tried to communicate with Robotis RS-485 servos within one of our projects at airq-ad. Below is our solution with explanation, tested on Cube Black, for anyone who might still need this.

First of all, according to STM32 reference manual:

"_If the RTS flow control is enabled (RTSE=1), then RTS is asserted (tied low) as long as the
USART receiver is ready to receive a new data. When the receive register is full, RTS is
deasserted, indicating that the transmission is expected to stop at the end of the current
frame._"

so I would say your "Expected behavior" is wrongly defined, i.e. RTS is high only when STM32 is not able to receive data any more. Therefore, "Hardware flow control mode" cannot be used for enabling RS-485 driver. STM32 also supports "Single-wire half-duplex communication" by automatically releasing TX wire when not transmitting, but without additional "Driver Enable" (DE) pin, it cannot be used for driving RS-485.

To conclude, RS-485 is not supported by hardware, but "Driver Enable" signal can be controlled from software, as described in ST Application note AN3070 - "Managing the Driver Enable signal for RS-485 and IO-Link communications with the STM32™’s USART". The idea is to use "Transmit complete interrupt" to change DE pin state. Indeed, NuttX supports USART interrupt based driving for DE pin (although it does not support DMA interrupt based driving).

For example, to enable RS-485 on Cube Black TELEM2 port,

1. run "make px4_fmu-v3_default qconfig" (you have to install qconfig first)
2. uncheck "System Type -> U[S]ART Configuration -> USART3 Rx DMA"
3. check "System Type -> U[S]ART Configuration -> RS-485 on USART3" and save the configuration
4. add "#define GPIO_USART3_RS485_DIR (GPIO_OUTPUT | GPIO_OPENDRAIN | GPIO_SPEED_50MHz | GPIO_PORTD | GPIO_PIN12 | GPIO_OUTPUT_CLEAR)" to boards/px4/fmu-v3/nuttx-config/include/boards.h
5. make clean + make px4_fmu-v3_default upload

You have to open the serial port with hardware flow OFF (_uart_config.c_cflag &= ~CRTSCTS;_). Then your TX (yellow) and RTS (blue) signals should look like this when transmitting:

[image]

Using the software interrupt method introduces additional latency, which may limit the maximum baud rate in your application. "Driver Enable" pin can be any free pin, we chose RTS pin for convenience. For other hardware or port, you'll have to change the config file / USART number / PIN number.

TomRvr, 2021-09-06:
Thanks a lot for those explanations !
I hope that I will be able to give it another try.

davids5, 2021-09-07:
@TomRvr Sorry I am late to the party. It looks like @TomRvr gave you the solution. The only comment I have is to check your HW and possibly use GPIO_PUSHPULL in the place of GPIO_OPENDRAIN:

in the `#define GPIO_USART3_RS485_DIR (GPIO_OUTPUT | GPIO_OPENDRAIN | GPIO_SPEED_50MHz | GPIO_PORTD | GPIO_PIN12 | GPIO_OUTPUT_CLEAR)"`

TomRvr, 2021-09-07:
Thanks @davids5 
I don't think I will have the time to try it soon
But if I do I'll post an update here.

klxing, 2022-10-28:
Hello everybody.
I also encountered the problem to commonicate with rs485 servo,but when I configured the marcos as this discussion described, the code return -1 when the function of write() is called.
here is how I write my code:
    char senddata[3] = {0xE8,0x02,0xBC};
    unsigned char buffer[16];
    double dist;
    int tx_ret = 0;
    PX4_INFO("fd_uart_read: %d\n",uart_read);
    tx_ret = write(uart_read,senddata,3);
    PX4_INFO("write_ret: %d\n",tx_ret);
where uart_read is the fd value returned by open() function. I have no idea why it will return -1when I call wrtie()

Can sb help me out or show me  some example of how to read and write data in application software to communicate through rs485?
Thanks a lot in advance!

davids5, 2022-10-28:
> tx_ret = write(uart_read,senddata,3);

Do you mean?
tx_ret = write(fd_uart_read, senddata, 3);

Please post the complete.

klxing, 2022-10-28:
yes david, i just make a assignment when transmit the parameter
here is my complete function:

int rs485_check(int uart_read)
{
    char senddata[3] = {0xE8,0x02,0xBC};
    unsigned char buffer[16];
    double dist;
    int tx_ret = 0;
    PX4_INFO("fd_uart_read: %d\n",uart_read);
    tx_ret = write(uart_read,0,1);
    tx_ret = write(uart_read,senddata,3);
    PX4_INFO("write_ret: %d\n",tx_ret);
    if(tx_ret > 0)//
        {
            printf("write first successful\n");
		    usleep(50);
            if(read(uart_read,&buffer,2) > 0)// read to buffer, 2 byte at one time
		    {
            printf("read first successful\n");
			PX4_INFO("Raw Data[0]: %x\n",buffer[0]);
			PX4_INFO("Raw Data[1]: %x\n",buffer[1]);
			dist = double(buffer[1]<<8|buffer[0]);
            printf("%f ",dist);
            printf("\n");
            }
        }
        else
        {
            PX4_INFO("Write Failed!");
        }
	usleep(5000);
	return 0;
}

And the  "tx_ret" will return -1

I was wondering whether it is because I have just wired RX and TX and leaves RTS unused. 

Or because the 5V power pin of the ttl to RS485 module  has insufficient voltage supply? But I was told that the wire has noting todo with the result of write() function because for serial ports, they can write the ttl even if you leave the pins unwired. I just cannot tell whetherr this is the truth.
 
I will test again tomorrom and will let you know if I get some result.

Thanks a lot!

davids5, 2022-10-28:
Where do you open the the file descriptor? 
Also write takes an fd, buffer and length  

`ssize_t write(int fildes, const void *buf, size_t nbyte);`
So what it is?
`tx_ret = write(uart_read,0,1);`

klxing, 2022-10-28:
> Where do you open the the file descriptor? Also write takes an fd, buffer and length
> 
> `ssize_t write(int fildes, const void *buf, size_t nbyte);` So what it is? `tx_ret = write(uart_read,0,1);`

I use this to get fd:

 int serial_fd = open(uart_name, O_RDWR | O_NOCTTY);

and i make : _uart_read = serial_fd_;  later
 and then pass _uart_read_ to write()  when calling:
 tx_ret = write(uart_read,senddata,3);

tx_ret = write(uart_read,0,1); is actually a test sentence I added to check if "tx_ret=-1" is caused by wrong input  when  I call tx_ret = write(uart_read,senddata,3);, and it should have been commented out. It should not have been there ,just added to do double check.

klxing, 2022-10-30:
> Where do you open the the file descriptor? Also write takes an fd, buffer and length
> 
> `ssize_t write(int fildes, const void *buf, size_t nbyte);` So what it is? `tx_ret = write(uart_read,0,1);`

Hey david,I want to conclude & update what I have done at this moment:
1.I set the configurations as mentioned by [Igor-AirQ](https://github.com/Igor-AirQ) by "make px4-fmuv2_default menuconfig":
uncheck "System Type -> U[S]ART Configuration -> USART3 Rx DMA"
check "System Type -> U[S]ART Configuration -> RS-485 on USART3" and save the configuration
2. Added "#define GPIO_USART3_RS485_DIR (GPIO_OUTPUT | GPIO_OPENDRAIN | GPIO_SPEED_50MHz | GPIO_PORTD | GPIO_PIN12 | GPIO_OUTPUT_CLEAR)" to boards/px4/fmu-v2/nuttx-config/include/boards.h
3. make my firmware and uploaded to pixhawk.  In my firmware the module inculdes the code of reading and writing rs485 data.
4. wired the pixhawk, the TTL to RS485 converter, a RS485 to USB converter ,and to my laptop through usb so I can check the data in serial communation tool. In the serial tool, I select bdrate= 115200, data Bits = 8, Parity =none, stop Bits =1, Flow Type = RTS/CTS.
5. start my module in QGC, which includes the read() and write() function.
6. After above process, no data can be seen in the serial tool software. Then I stop my module, type: "mavlink start -d /dev/ttyS2 -b 115200 -z -x", and start my module again.
7. This time in the serial tool software I can see data floods in. However, the data just makes no sense. And in my module the wrtie() and read() function returned -1, which means these data is not sent/received by my code.
  Just cannot figure where is wrong. I think there must be some mistake in my configuration or there's some interference  I didn't notice. Can you check my configuration? Thanks a lot!

P.S., the code in my module related to read/write RS485 are(I am not listing all the codes, just post the lines I think have sth to do with the serial port configuration):

        char rs_path[] = "/dev/ttyS2";
       speed = 115200;
       fd_uart_read = open(uart_name, O_RDWR | O_NOCTTY);    	
	set_bd = set_uart_baudrate(fd_uart_read ,speed);
        uart_config.c_oflag &= ~ONLCR;
        uart_config.c_cflag &= ~CRTSCTS;
       uart_config.c_cflag &= ~(CSTOPB | PARENB);
       termios_state = cfsetispeed(&uart_config, speed);
       termios_state = cfsetospeed(&uart_config, speed);
       termios_state = tcsetattr(fd_uart_read, TCSANOW, &uart_config);

     char senddata[3] = {0xE8,0x02,0xBC};
    unsigned char buffer[16];
    double dist;
    int tx_ret = 0;
    tx_ret = write(fd_uart_read,senddata,3);
    PX4_INFO("write_ret: %d\n",tx_ret);
   usleep(50);
   if(read(fd_uart_read,&buffer,2) > 0)
    {
            printf("read first successful\n");
	PX4_INFO("Raw Data[0]: %x\n",buffer[0]);
	PX4_INFO("Raw Data[1]: %x\n",buffer[1]);
	dist = double(buffer[1]<<8|buffer[0]);
         PX4_INFO("%f \n",dist);
    }
     else
       {
            PX4_INFO("read failed!");
       }

davids5, 2022-10-31:
@klxing 

After the call to open print the value of fd_uart_read - What is it?
Is the open call on the same task that the write is?  (the open has to be in the same task as the users of the FD)

Do you have a pull up on the DIR Line? If not GPIO_OPENDRAIN  needs to be GPIO_PUSHPULL.

klxing, 2022-10-31:
> 

that's because I want to check if it was opened properly and remains the same. And they are both eauals to 3.   the open() and users of FD is in the same task, just different functions.
 GPIO_OPENDRAIN and GPIO_PUSHPULL have been tried, it jusy return ret=-1: "Error writing: Unknown error"
driving me crazy

davids5, 2022-10-31:
I would debug it with Jtag.

klxing, 2022-10-31:
> I would debug it with Jtag.

yes i relalized that I should learn how to use Jtag.

Now I switched to USART2(TELEM1), and I can see data floods in my serial tool, which means pixhawk is transmitting data, but the data are not what I set, and my function still report error in read(）and write() ……

klxing, 2022-11-02:
> @TomRvr Sorry I am late to the party. It looks like @TomRvr gave you the solution. The only comment I have is to check your HW and possibly use GPIO_PUSHPULL in the place of GPIO_OPENDRAIN:
Hey David, one more question please
I see the configuration of " tty.c_iflag &= ~(IXON | IXOFF | IXANY)" to disable software of the port, is this necessary? I am not sure whether the failure has sonething to do with this or not.
And ,is there some command that I can check whether the port(dev/ttyS2) is occupied by other tasks? I also doubt that now but havn't found the way to check it.

davids5, 2022-11-02:
@klxing use ps and look for mavlink using that telem port

klxing, 2022-11-03:
> @klxing use ps and look for mavlink using that telem port

Thanks  @David.

I tried typing "ps" in the nsh but it says:"ps: command not found"...as I googled some suggest doing "mount -t procfs /proc" but it turns out to be unhelpful...

Also, I found one interesting thing. As there is a "serial_test" command in the project, when I use this command to read or write some byte separetly, it will success, but when doing read and write at the same time, reading bytes will be wrong. The 
 source file(serial_test.c) of this command seems using the same write() function as I did in my module , still just cannot figure out why my write() function will return errno=22 and read() errno=13.

klxing, 2022-11-03:
> @klxing use ps and look for mavlink using that telem port

@davids5 Sorry for so many disturbance.
After all these suffering in trying to read rs485 data, it turns out that it's just because I made a really stupid mistake.
I put open() and init() function of rs485 in the Module::init() while read() and write() function in Module::Run()
This causes the mistake when calling read() and write(). Now when I put open(), init() , read() and write() all in Module::Run(), it worked as desired.
Thanks for your kind help!
There may be nobody made such mistake as I did but I want to draw this issue to a conclusion.

Also, I tested that using some ttl to 485 converters that don't require the signal of RTS control, the pin can be left blank, it works well for me.

davids5, 2022-11-03:
Glad you found it.

closed 2022-11-03 by davids5