PX4/PX4-Autopilot #16194: Cube Orange icm20649 causes poor position tracking

aet2230, 2020-11-15:
General:
We use quadrotor in laboratory environment (motion capture).
Using Pixhawk2 (cube black) we have very good position tracking. After replacing Pixhawk2 (on the same quadrotor) by CubeOrange, the performance deteriorated (from +/- 1[cm] to +/- 5[cm]). This due to lower degree of belief of ekf2 to VO (this may be observed on the attached uLog plots of position and EKF innovations)

[image]
[image]
[image]
[image]

After long investigation and trials we decided to disable one IMU each time (hard coded in rc.board_sensors). We found that disabling icm20649 completely solved the problem. Also the cube temperature decreased (it was not very hot before, but now it's temperature after flight is similar to pixhawk2's)
Remarks: 
1) we did not observe any excessive IMU noise
[image]
[image]

2) we use the following setup and parameters:

Quadrotor with motion capture as the only positioning and yaw sensor

EKF2_AID_MASK	24
EKF2_EVV_NOISE	0.10
EKF2_HGT_MODE	3
EKF2_MAG_TYPE	6

dagar, 2020-11-15:
Did you see similar results between the icm20948 and icm20649? The relevant difference in the IMU is the dynamic range (16G vs 30G) and where they're mountd in the Cube. 

The other thing that might be useful in this case is trying multi-EKF where you run an instance of EKF2 for each IMU. The Cube Orange (stm32h7) can easily afford this.

dagar, 2020-11-15:
For now we can start the icm20602 first. https://github.com/PX4/PX4-Autopilot/pull/16195

aet2230, 2020-11-15:
> Did you see similar results between the icm20948 and icm20649? The relevant difference in the IMU is the dynamic range (16G vs 30G) and where they're mountd in the Cube.
> 
> The other thing that might be useful in this case is trying multi-EKF where you run an instance of EKF2 for each IMU. The Cube Orange (stm32h7) can easily afford this.
> 
> ```
>         # enable multi-ekf
> 	param set EKF2_MULTI_IMU 3
> 	param set SENS_IMU_MODE 0
> ```

No, we disabled all IMUs and started one each time. We found that when icm20602 or icm20948 or both were enabled, the performance was good. Didn't try MULTI_IMU (it is in master, not in stable v1.11.1 version) and did not try to change the order of imu starting

dagar, 2020-11-15:
> No, we disabled all IMUs and started one each time. We found that when icm20602 or icm20948 or both were enabled, the performance was good. Didn't try MULTI_IMU (it is in master, not in stable v1.11.1 version) and did not try to change the order of imu starting

It's probably the vibration isolation. We should have been starting the isolated SPI4 sensors first rather than arbitrarily in bus numbering order. https://github.com/PX4/PX4-Autopilot/blob/053bf7e732ecbd3732cc3a24d64a069ec2d2bff4/boards/cubepilot/cubeorange/init/rc.board_sensors#L11-L14

https://github.com/PX4/PX4-Autopilot/pull/16195

In the future it will be interesting to see the side-by-side estimator data in the same log.

dagar, 2020-11-16:
Should be fixed by https://github.com/PX4/PX4-Autopilot/pull/16195.

closed 2020-11-16 by dagar

RohitBostin, 2023-03-17:
@aet2230 iam facing the same problem can you tell me how you disabled icm20649

aet2230, 2023-03-17:
Go to 
/boards/cubepilot/cubeorange/init/rc.board_sensors

Comment / delete the following line:
icm20649 -s -b 1 start