PX4/PX4-Autopilot #16200: FAILED: make px4_sitl gazebo

joker-csee, 2020-11-16:
Hi there,
I'm using Ubuntu 20.04 with gazebo 11.2.0, errors occurred when make px4_sitl gazebo. How can I deal with it?

$ make px4_sitl gazebo
-- PX4 version: v1.11.0-rc3-694-g053bf7e732
-- PX4 config file: /home/d207/Documents/PX4-Autopilot/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.5", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-470-gda9f314
-- ROMFS: px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /home/d207/Documents/PX4-Autopilot/build/px4_sitl_default
[0/770] git submodule src/drivers/gps/devices
[8/770] git submodule src/lib/ecl
[10/770] git submodule mavlink/include/mavlink/v2.0
[15/770] git submodule Tools/sitl_gazebo
[21/770] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 standard
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components: system thread filesystem 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 2.88
[... 107 lines cut ...]
    linked by target "gazebo_uuv_plugin" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "nav_msgs" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_sonar_plugin" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_vision_plugin" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_gst_camera_plugin" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "mav_msgs" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_controller_interface" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_gimbal_controller_plugin" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_imu_plugin" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_mavlink_interface" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_motor_model" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_magnetometer_plugin" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_barometer_plugin" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "LiftDragPlugin" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_catapult_plugin" in directory /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo

-- Configuring incomplete, errors occurred!
See also "/home/d207/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/d207/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[761/770] Linking CXX static library src/systemcmds/tests/libsystemcmds__tests.a
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/d207/Documents/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -GNinja /home/d207/Documents/PX4-Autopilot/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/d207/Documents/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[764/770] Building CXX object src/syst..._listener.dir/listener_generated.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:226: px4_sitl] Error 1

closed 2020-11-25 by joker-csee

adbidwai, 2021-06-13:
@joker-csee How did you resolve this error ?

joker-csee, 2021-06-13:
> @joker-csee How did you resolve this error ?
try to uninstall gazebo by apt remove, and reinstall gazebo by apt install ros-noetic-gazebo.