PX4/PX4-Autopilot #16213: questions regarding "cmake build config"

Daniel900725, 2020-11-17:
The problem looks like the following:

[image]

I followed the instruction from the website: https://dev.px4.io/master/en/setup/vscode.html
The Visual Studio keep saying that "CMAKE execution file has a problem and did you install it and did cmake.cmakePath included properly?" I installed the CMAKE file following the instructions.

Could you please help on this issue? 

I appreciate your help a lot in advance!

dagar, 2020-11-17:
What's your development environment? Do you have the toolchain installed and working (https://dev.px4.io/master/en/setup/dev_env_linux.html)?

Daniel900725, 2020-11-18:
Thank you for your reply.

I have downloaded Cygwin Toolchain as the PX4 Development environment on
Windows.
Do I have to use Linux?

Thank you in advance

Best Regards

Daniel Kim

dagar, 2020-11-18:
> Do I have to use Linux?

No, but I would strongly recommend it. Using Ubuntu within Windows (WSL or WSL2) works quite well these days, including from within VSCode.

Daniel900725, 2020-11-18:
“CMake-related issue” shown in the photo can be solved if I use other tool
chain?
I dont have concept of tool chain. Is it one of compilers?
I think that complier is not working for my case.

thank you in advance.

dagar, 2020-11-18:
In this context toolchain refers to arm-none-eabi-gcc. 

This is what I'd recommend you use on Windows, it's much faster than cygwin without most of the headaches.
https://code.visualstudio.com/blogs/2019/09/03/wsl2

Within WSL (Linux on Windows) you install the normal PX4 development environment. https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html

Daniel900725, 2020-11-19:
Thank you for your reply,

Here is my current situation and a few questions:

1. I have no experience of using LINUX yet so if possible, could you please
suggest an alternative way which does not require me to use LINUX such as
Windows only?

2. I have downloaded the PX4 source code version 1.9.0 and tried to build
following the instructions from the website:
https://dev.px4.io/master/en/setup/vscode.html but I cannot build the
source.
The box on the bottom of the right side pops up. It says that CMAKE
execution fille or cmake.cmakepath issue. How can I solve this problem?
[image: 2.PNG]

3. Regarding PX4-detect, it says that PX4 detect has not been selected. I
think that the compiler has not been connected in the IDE.
If this issue is caused by the absence of a compiler, how can I connect
manually?
[image: 3.jpg]

I really appreciate your help.

Daniel Kim