PX4/PX4-Autopilot #16223: stm32h743xih6 Automatic restart （More than 4 restarts of 500 aircraft in three hours）

MURANzhang, 2020-11-18:
**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/master/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

MURANzhang, 2020-11-18:
1. At the beginning, the error was prompted that the static local variable of _spi_bus4 was overwritten, and the value of the inexplicable general register R5 was changed, and then I gave a constant address directly;
2. Afterwards, the error occurred when the stm32_recvfifo interface was called. The error report phenomenon is also that the value of general register R1 is suddenly changed, there is no ldr and str operations, no stack information, user stack and program stack are not stepped on;
3. In the error report, the value of the CFSR register is 0x00000400, and the value of the HFSR register is 0x40000000;

MURANzhang, 2020-11-18:
Searching for information found that the currently running M7 kernel prompts an inaccurate error, and BFSR sets the flag bit, and it is impossible to accurately find the code where the exception occurred through the memory and stack.

dagar, 2020-11-18:
What board is this and what version of PX4 is it running?

MURANzhang, 2020-11-18:
> 这是什么板子，正在运行什么版本的PX4？
stm32h743xih6

MURANzhang, 2020-11-18:
board：STM32H743XIH6
from px4 branch: 6e18cb85b1911ac052838c73a6fbb9b8d6f72d2c

The problem I have now is that through the hardfault information record, I saw that the M7 kernel concluded that this is an inaccurate error. 
There is no way to trace the actual location of the exception through the stack information.
 I guess it is because the memory has been changed. Error.
do you have any advanced debugging ideas?

18235655008@163.com
 
From: Daniel Agar
Date: 2020-11-18 23:02
To: PX4/PX4-Autopilot
CC: MURANzhang; Author
Subject: Re: [PX4/PX4-Autopilot] stm32h743xih6 Automatic restart (#16223)
What board is this and what version of PX4 is it running?