PX4/PX4-Autopilot #16225: SITL MAVROS rover mission test failing intermittantly

dagar, 2020-11-18:
https://github.com/PX4/PX4-Autopilot/pull/16217/checks?check_run_id=1415920793

[image]

Jaeyoung-Lim, 2020-11-28:
Looking closer, it is simply the test failing.
```
[ROSTEST]-----------------------------------------------------------------------

[px4.rosunit-rover_mission_1/test_mission][FAILURE]-----------------------------
position not reached | lat: 47.397737805, lon: 8.546049032, alt: 488.02, current pos_xy_d: 34.27, current pos_z_d: 0.05, best pos_xy_d: 34.23, best pos_z_d: 0.00, index: 0 | timeout(seconds): 60
```

dagar, 2020-11-30:
> Looking closer, it is simply the test failing.
>
> ```
> [ROSTEST]----------------------------------------------------------------------- [...]

That's better than a segfault. I'll take a closer look at the log next time I see it.

chfriedrich98, 2025-02-11:
Closing this issue since this test no longer seems to fail.

closed 2025-02-11 by chfriedrich98